mvp/uhubctl #169: Anker A7516 works with uhubctl

noahp, 2019-08-14:
Verified this port works (with an attached USB2 peripheral, hub plugged into a USB3 root hub):
https://www.amazon.com/gp/product/B00XMD7KPU/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Ouptut from `uhubctl`:
```plaintext
Current status for hub 1-1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

mvp, 2019-08-14:
Thanks for your report!
Did you check for VBUS turn off per https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl?
If VBUS off is not working, we don't add hubs into supported list. Thanks!

noahp, 2019-08-15:
Ah, you're correct, it caused an attached device reenumeration but did not cycle VBUS, sorry about that! Closing.

closed 2019-08-15 by noahp