mvp/uhubctl #173: Power off not equivalent to plug removal

emsaks, 2019-09-06:
I'm using an AmazonBasics 4-port hub (p/v id: 2109:2811) that is listed as working with `uhubctl`.
In general it seems to work, but when cycling power to a hard drive, I see a difference between physical removal and just cycling. 
I have a hard drive that slows after a minute and have to reset it by cycling the power. If I use `uhubctl` I can see that the drive is properly reset (`dmesg` shows that the drive is spinning up etc.) but it doesn't speed up the drive - only if I physically remove the plug. I've tried using a delay before repowering on, tried unbinding the driver and setting `pm_qos_no_power_off` to `0` with no luck.

mvp, 2019-09-06:
Can you check that your hub turns off VBUS? Follow this FAQ item: https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl
Also, make sure you are using USB3 upstream connection and USB compliant cable - hub should be detected as 2 devices: USB3 and USB2.

emsaks, 2019-09-07:
I'll check out the vbus, but it is listed as supported on the site. Re.
USB3, it's being used on a PC that only supports usb2.

mvp, 2019-09-07:
There are many possible configurations, and many device revisions. Some revisions use different USB chips which don't work - e.g. D-Link DUB-H7 rev A works, when rev B doesn't, rev C works weird, and rev D works good once again.

AmazonBasics hub was reported as working by multiple users, but few had issues. We don't know if this was due to USB 2 vs USB 3 connection. There is also known incompatibility of AmazonBasics hub with raspberry pi.

As far as we know, AmazonBasics hub works well when connected as USB 3 device.

mvp, 2019-09-12:
Closing for inactivity. Please comment here or feel free to reopen if you have anything to report. Thanks!

closed 2019-09-12 by mvp