mvp/uhubctl #175: Different libusb_bos_descriptor in FreeBSD 11.2

Cirromulus, 2019-09-20:
Hi there!
I was trying to compile uhubctl on a freeNAS jail.
Build fails however, because `bNumDeviceCaps` is not a member of `struct libusb_bos_descriptor`.

My libusb.h 
```
$FreeBSD: releng/11.2/lib/libusb/libusb.h 331722 2018-03-29 02:50:57Z eadler 
[...]
typedef struct libusb_bos_descriptor {
        uint8_t bLength;
        uint8_t bDescriptorType;
        uint16_t wTotalLength;
        uint8_t bNumDeviceCapabilities;
        struct libusb_usb_2_0_device_capability_descriptor *usb_2_0_ext_cap;
        struct libusb_ss_usb_device_capability_descriptor *ss_usb_cap;
        struct libusb_bos_dev_capability_descriptor **dev_capability;
}       libusb_bos_descriptor __aligned(sizeof(void *));
```

Changing the line in uhubctl.c:432 from `bNumDeviceCaps` to `bNumDeviceCapabilities` fixed the problem.

However, it does not find any smart hubs; but this may just be my cheap mainboard missing the capability.

mvp, 2019-09-20:
Fixed in 4818edd.

mvp, 2019-09-20:
Thanks for your submission! :-)

closed 2019-09-20 by mvp