mvp/uhubctl #177: Coolgear USBG-4U3MLR2 works

mikepurvis, 2019-09-26:
Probably other Coolgear models as well, but this is the one I have and checked: https://www.coolgear.com/product/usb-3-1-4-port-metal-hub-wdin-rail-clip-esd-surge-protection

mvp, 2019-09-26:
Thanks for reporting!
Did you check for VBUS off support? https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl

2. Check for VBUS (voltage) off support: plug a phone, USB light or USB fan into USB port of your hub. Try using uhubctl to turn power off on that port, and check that phone stops charging, USB light stops shining or USB fan stops spinning. If VBUS doesn't turn off, your hub manufacturer did not include circuitry to actually cut power off. Such hub would still work to cut off USB data connection, but it cannot turn off power, and we do not consider this supported device.

mikepurvis, 2019-09-26:
Yes it does— I plugged in an STM32 Discovery Kit, and the main 5V power LED goes off when commanded to do so.

mikepurvis, 2019-09-26:
FWIW I can't actually use uhubctl in my project due to the GPL license, but it was helpful to verify that my hub is compatible before building my own implementation.

mvp, 2019-09-27:
Thanks, I will add this hub to compatible list. Regarding license, it seems that these days uhubctl is present in package repositories of most popular Linux distributions, e.g. Ubuntu. I don't see why doing as simple as `sudo apt install uhubctl` would be a licensing problem.

mvp, 2019-09-27:
Added in 6e5782c.

closed 2019-09-27 by mvp

mikepurvis, 2019-09-30:
Here's the ID string if you want the VID/PID as well:

```
05e3:0620 GenesysLogic USB3.1 Hub, USB 3.10, 4 ports
```

mvp, 2019-10-01:
Updated this info in 54365acb8db.

mikepurvis, 2019-10-15:
Uuuugh, I'm really sorry about this, but I've done further testing on this and it appears that this hub actually does not work, or is at best inconsistent.

I don't know why/how it appeared to work previously, but it definitely doesn't in my testing now. I promise I was thorough before— I'm using the exact same stm32 disco kit, and I now can't get it to turn off.

Sorry again for the misinformation.

mvp, 2019-10-15:
Does uhubctl turn off power for both USB3 and USB2 counterpart hubs? In some connection scenarios (e.g. when connected via RPi4) uhubctl gets confused about finding both and only turns power off on one. However, USB3 hubs require to turn off power on both for this to actually work.

You can use `uhubctl -e` twice to turn off power for those hubs separately. Can you please try that? Thanks

mikepurvis, 2019-10-16:
Yeah, tried both separately and together, and with both a USB2 device connected (the STM32 board) and a USB3 device (another bus-powered hub with an LED on it). Neither device goes dark. Also no change if I change the internal jumper of the hub to disable bus-powered mode. So I really don't know what I did that persuaded me previously that this was working. :(

mvp, 2019-10-17:
That's unfortunate. Oh well, I will remove it from the list then :-(

mvp, 2019-10-17:
Removed Coolgear USBG-4U3MLR2 from supported list in fd611df.

liljenstolpe, 2022-03-21:
I have tested a similar coolgear.  It does not switch the VBUS.  It does switch off the data.  Actually doing an -s for the device I want to toggle lists both the USB2 and USB3 ports (even though the device is only on the USB2 bus) - so the port pairing is working.  However, I did try with the -e and toggling both busses - VBUS is still hot.