mvp/uhubctl #178: D-Link DUB-H7 rev C1 and C2 (black edition)

acubino, 2019-09-27:
Dear.

I just bougth two units of DLink DUB-H7 (black edition) one is rev C1 and another rev C2. Both have same usb ids:
Bus 003 Device 016: ID 0835:8502 Action Star Enterprise Co., Ltd 
Bus 003 Device 015: ID 0835:8500 Action Star Enterprise Co., Ltd 
Bus 003 Device 014: ID 0835:8501 Action Star Enterprise Co., Ltd 
Bus 003 Device 013: ID 0835:8500 Action Star Enterprise Co., Ltd 

Both are not listed using ./uhubctl.

With lsusb -v can see the following wHubCharacteristic 0x00e4 that is not listed in main page as incompatible (but not as compatible too).

 wHubCharacteristic 0x00e4
    Ganged power switching
    Compound device
    Ganged overcurrent protection
    TT think time 32 FS bits
    Port indicators

Test in Armbian, also in Ubuntu 18.04 with compiled last version.

Full lsusb -v attached.
[lsusb.txt](https://github.com/mvp/uhubctl/files/3663593/lsusb.txt)

mvp, 2019-09-27:
Yes, unfortunately, you are correct D-Link DUB-H7 only rev A and D are working and listed in compatibility list. Rev B and C do not support power switching.

closed 2019-09-27 by mvp

acubino, 2019-09-27:
Maybe add in the list of supported
* rev B and C not support power switching?.

because i suppose that all models works but only untested (yes, is my fault).