mvp/uhubctl #180: Raspberry Pi 3 Model B+ Port keeps coming up after disabling

aprin81, 2019-10-08:
Hi, anyone encountering the same issue?
this is the setup:
```
pi@rpi-1:~ $ uhubctl
Current status for hub 1-1.3 [0424:2512]
...
  Current status for hub 1-1.2 [0424:2512]
...
Current status for hub 1-1.1 [0424:2514]
  Port 1: 0503 power highspeed enable connect [0424:7800]
  Port 2: 0100 power
  Port 3: 0100 power
Current status for hub 1-1 [0424:2514]
  Port 1: 0503 power highspeed enable connect [0424:2514]
  Port 2: 0503 power highspeed enable connect [0424:2512]
  Port 3: 0503 power highspeed enable connect [0424:2512]
  Port 4: 0100 power
```
disabling here:
```
pi@worker11:~ $ sudo uhubctl -a off -p 2 -l 1-1
Current status for hub 1-1 [0424:2514]
  Port 2: 0503 power highspeed enable connect [0424:2512]
Sent power off request
New status for hub 1-1 [0424:2514]
  Port 2: 0000 off
```

after 5-10 seconds:
```
Current status for hub 1-1 [0424:2514]
  Port 1: 0503 power highspeed enable connect [0424:2514]
  Port 2: 0503 power highspeed enable connect [0424:2512]
  Port 3: 0503 power highspeed enable connect [0424:2512]
  Port 4: 0100 power
```
I've tried different combinations and extra disable commands to other ports but it persists.
Any ideas?

mvp, 2019-10-08:
This is FAQ item. Try adding `-r 100`.

closed 2019-10-08 by mvp

aprin81, 2019-10-08:
Thanks, i focused on the RPi-specific paragraph since i m using different versions
I ll try that tomorrow :)