mvp/uhubctl #181: Plugable USB2-HUB10S not compatible with uhubctl

tony-wallace, 2019-10-08:
The README.md says that the Plugable USB2-HUB10S 10-port USB 2 Hub is compatible. 
I bought one used from ebay. 
It does not seem to be compatible on my Ubuntu 18.04 Dell Precision 5820
* sudo uhubctl does not report it. 
* sudo lsusb -v does not claim Per-port power switching
  * output: https://gist.github.com/tony-wallace/8b45a82fc745ceb3dfe4e889475abc45

Incidentally, uhubctl does report some internal USB hubs/ports on this Dell, and lsusb -v agrees those hubs claim per-port power switching. Uhubctl reports successfully turning power on and off on those ports (though it is confused about which Hub I selected- it always hits both Hubs).  However, I never did get the power to actually turn off.  I chalked that up to incompatible hardware and sent away to ebay for the Plugable USB2-HUB10S because I wanted a 10-port USB 2.0 hub.  

It looks like this Plugable USB2-HUB10S is functional in other respects.

mvp, 2019-10-09:
Thanks for your detailed report, I have removed this hub from supported list in a6cdc3d.

> Uhubctl reports successfully turning power on and off on those ports (though it is confused about which Hub I selected- it always hits both Hubs)

If your hubs are USB3, this is expected - uhubctl tries hard to control power for both USB3 hub and its USB2 compatibility counterpart hub unless you disable this with `-e` - this is required to actually turn off power for USB3.
If your hubs are USB2 only, this should never happen - I would like to see full output of `sudo uhubctl` and uhubctl version (please use 2.1.0 or later).

There is certainly possibility that your onboard hubs don't support VBUS despite claiming support, and this is manufacturer's fault :-(

closed 2019-10-09 by mvp

tony-wallace, 2019-10-09:
Thanks for the reply. 
FYI, I re-ran the lsusb and uhubctl on the same workstation, with and without the Plugable USB2-HUB10S connected. 

* With the hub connected: https://gist.github.com/tony-wallace/1fcf4f8067ec1349632fe78f054385e2
* Without the hub connected: https://gist.github.com/tony-wallace/81c4807b93cd79fc695c4a2beca1e528

tony-wallace, 2019-10-09:
Still have faith in uhubctl. Based on README file change history, I just ordered a rev-D2 DLink DUB-H7 from ebay. Hope it is a rev-D2 anyway!  It said so on the label shown in the pic.

mvp, 2019-10-10:
Please report if D-Link DUB-H7 rev D is working for you or not. If it doesn't, older D-Link DUB-H7 rev A (silver) is definitely working perfectly - I am using it myself, and it seems to be relatively easy to find on eBay right now under $10.

tony-wallace, 2019-10-15:
My D-Link DUB-H7 rev D2 definitely works with uhubctl.  Cheers!