mvp/uhubctl #182: Transcend USB 3.0 Hub 4 Port Powered Hub3 working

pilaGit, 2019-10-14:
Transcend USB 3 Hub3

Hub has 4 USB 3.0 ports. Only the single USB port on the narrow side can be controlled. Remaining 3 are ignoring settings. Tested on a Raspberry Pi 4 with LibreElec 9. Take care: there is almost identical HUB2 which does not have external Power Supply. I am talking about the HUB3 with external power supply. Back is labeled: TS-HUB3K. Power supply is declared at 18W. Mine were made in 2016.

```
rpi4:~ # lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 010: ID 8564:4000 Transcend Information, Inc. RDF8
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

rpi4:~ # ./uhubctl
Current status for hub 2 [1d6b:0003 Linux 4.19.66 xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.3 [8564:4000 Transcend USB2.0 Hub]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.66 xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]

```
# USB port at 1-1.3 (only the single separate USB port on the narrow side turns off and on)
./uhubctl -l 1-1.3 -a off -p 1
./uhubctl -l 1-1.3 -a on -p 1

mvp, 2019-10-14:
This hub Transcend TS-HUB3K was already added https://github.com/mvp/uhubctl/issues/121 and then removed https://github.com/mvp/uhubctl/issues/135 from compatibility list before in , so we have to be very careful.

Your report is quite confusing - it does not seem like you have connected TS-HUB3K using compliant USB3 connection - I only see USB2 connection (we should test for both).

Can you please test it on desktop computer (Linux or Mac) with USB3 upstream connection? You should see this hub advertising itself as 2 hubs: one USB3, and another USB2 compatibility counterpart.
Then test all ports if they actually turn off VBUS or not as described in FAQ: https://github.com/mvp/uhubctl/blob/master/README.md#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl
Thanks!

pilaGit, 2019-10-14:
I was incomplete, sorry. My statement was true if not completely documented: it only works on one port when connected to USB2 device. When connected to USB 3 port on the same RPi4, no ports can be turned off. 

RPi4 USB3 + Transcend USB 3 Hub
rpi4:~ # lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 8564:4000 Transcend Information, Inc. RDF8
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 013: ID 8564:4000 Transcend Information, Inc. RDF8
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

rpi4:~ # ./uhubctl
Current status for hub 2-1 [8564:4000 Transcend USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2 [1d6b:0003 Linux 4.19.66 xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [8564:4000 Transcend USB3.0 Hub, USB 3.00, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.1 [8564:4000 Transcend USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.66 xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]

# turn off USB 2 ports 1-4 at 1-1.1 does not achieve anything
./uhubctl -l 1-1.1 -a off -p 1

# turn off USB 3 ports 1-4 na  1-1.1 does not achieve anything, ends in an error
./uhubctl -l 2-1 -a off -p 1

No compatible smart hubs detected at location 2-1!

mvp, 2019-10-15:
Actually, this might be an issue with `uhubctl` mechanism of discovering USB2/3 hub pairs.
Current algorithm expects that both USB3 hub and its compatibility USB 2.1 partner have the same location depth, e.g. `2-1.2.3` for USB2 and `3-1.2.3` for USB3:
https://github.com/mvp/uhubctl/blob/4818eddc0389e2bba34a8cbe3ca5c67a6ad4d5ab/uhubctl.c#L753-L760
However, RPI4 has very weird USB3 configuration, because it allocates root USB3 and USB2 hubs at different depth.

In your example, your USB3 hub is `2-1`, and USB2 is `1-1.1`, which is not expected by current uhubctl code:

```
Current status for hub 2-1 [8564:4000 Transcend USB3.0 Hub, USB 3.00, 4 ports]
Port 1: 02a0 power 5gbps Rx.Detect
Port 2: 02a0 power 5gbps Rx.Detect
Port 3: 02a0 power 5gbps Rx.Detect
Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.1 [8564:4000 Transcend USB2.0 Hub, USB 2.10, 4 ports]
Port 1: 0100 power
Port 2: 0100 power
Port 3: 0100 power
Port 4: 0100 power
```

You should still be able to turn off power by using 2 commands when using USB3 connection, e.g.:

```
uhubctl -l 2-1 -a 0 -p 1    # turn off for USB3
uhubctl -l 1-1.1 -a 0 -p 1   # turn off for USB2
```
(You don't need this when using USB2 connection).

In meantime, I will consider changing `uhubctl` code to allow such weird configuration - thanks for your report!

mvp, 2019-10-17:
Relaxed level check when looking for USB2/3 dual hubs in 743ecf226 - this should make any hub connected to RPi4B behave better. It should not require using `-e` twice to turn power off - please check. Thanks!

mvp, 2019-10-17:
Can you please re-check Transcend TS-HUB3K again using this FAQ item

https://github.com/mvp/uhubctl/blob/743ecf226b72bbb/README.md#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl

You should  build uhubctl from master branch please. Thanks!

mvp, 2019-10-24:
Sorry, closing this due to inactivity from submitter. Please free to reopen if you check that this hub definitely supports VBUS off per FAQ item above. Thanks!

closed 2019-10-24 by mvp