mvp/uhubctl #183: Permission problems

kajzersoze, 2019-10-22:
I get this problem when running `uhubctl` as root 
```
No compatible smart hubs detected!
Run with -h to get usage info.
There were permission problems while accessing USB.
To fix this, run this tool as root using 'sudo uhubctl',
or add one or more udev rules like below
to file '/etc/udev/rules.d/52-usb.rules':
SUBSYSTEM=="usb", ATTR{idVendor}=="2001", MODE="0666"
then run 'sudo udevadm trigger --attr-match=subsystem=usb'
```

after doing what is suggested with udev rules I get the same thing.

I'm not using `sudo` btw but I'm logged in as root with `su`

How can I overcome this ?

mvp, 2019-10-22:
If you are already root, you don't need to use `sudo`.

Can you post output of `sudo lsusb -v`? Specifically, search for string `Per-port power switching`.
If you don't see it in `lsusb` output, your hub is not supported - see more in this FAQ item: https://github.com/mvp/uhubctl#what-is-usb-per-port-power-switching

That said, I still would like to understand why you are seeing permission problems even when you are root - post output of `sudo lsusb -v` please. Thanks!

kajzersoze, 2019-10-22:
[Here](http://dpaste.com/387QADB) is the output

I know I don't need sudo, I just mentioned that because I don't get it either (permissions problems), don't know what to say about that, hence asking about it :)

mvp, 2019-10-24:
It seems that you have some issues with your device descriptors, and this propagates to manifest itself as permission issue even for root.
Added workaround in d227c8326f3cbc248a7ae603e7347afe86e963ee - it will not complain about permissions if we are already running as root.

closed 2019-10-24 by mvp

kajzersoze, 2019-10-24:
Yeah that solved that, thanks.

Now I get this 
```
# uhubctl                                                                                                                                                                                                               
No compatible smart hubs detected!
Run with -h to get usage info.
```

Well, it is what it is.