mvp/uhubctl #186: Texas Instruments, Inc. TUSB2046 Works

cron410, 2019-11-07:
I'm able to turn off ports with -r 1000 -a 0

mvp, 2019-11-07:
Many hubs may _appear_ to work, but actually do not turn port voltage off.

Can you please check your hubs per FAQ:
https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl
> Check for VBUS (voltage) off support: plug a phone, USB light or USB fan into USB port of your hub. Try using uhubctl to turn power off on that port, and check that phone stops charging, USB light stops shining or USB fan stops spinning. If VBUS doesn't turn off, your hub manufacturer did not include circuitry to actually cut power off. Such hub would still work to cut off USB data connection, but it cannot turn off power, and we do not consider this supported device.

Also, please do this for all other hubs you submitted #187, #188, #189, #190.

Thanks for your submission!

mvp, 2019-11-11:
TUSB2046 is generic chip manufactured by Texas Instruments, and can be used by various hub vendors, and configured to support or not support per-port power switching. Please provide actual hub vendor and model.

closed 2019-11-11 by mvp