mvp/uhubctl #188: D-Link DUB-H4 rev B2 works

cron410, 2019-11-07:
I'm able to turn off ports with -r 1000 -a 0

mvp, 2019-11-07:
I highly doubt that this one actually works for power off.
D-Link DUB-H4 are known to work starting from revision D, not earlier.

cron410, 2019-11-07:
Mine's silver, i read the sticker to get this model number. It's an
externally powered unit.

mvp, 2019-11-11:
Please check if VBUS off is supported per https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl. We do not add hubs which don't support VBUS off.

cron410, 2019-11-12:
This absolutely works. I'm able to power down a USB fan, hard drive, and a USB flash drive with it. I use it to power down the microcontroller in a 3d printer remotely. It draws 5v power from the USB cable after I turn off the wifi plug that shuts down the main PSU of the printer. In the attached screenshot, only the hub is plugged into the computer, and the FTDI device is the 3d printer.

cron410, 2019-11-12:
```
Bus 002 Device 067: ID 05e3:0605 Genesys Logic, Inc. USB 2.0 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0605 USB 2.0 Hub
  bcdDevice            6.0b
  iManufacturer           0
  iProduct                1 USB2.0 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00e9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 32 FS bits
    Port indicators
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
```
[image]
[image]

mvp, 2019-11-13:
Much thanks for your detailed report. Added to the table

closed 2019-11-13 by mvp