mvp/uhubctl #191: Dell U2415 works

timocharis, 2019-11-12:
Dell UltraSharp U2415 24" LCD Monitor works

mvp, 2019-11-12:
Thanks for your report. Can you please check if vbus off is working? Connect your phone to this monitor, turn off that USB port using uhubctl and check if phone stops charging.

timocharis, 2019-11-12:
Yes it does. I'm still having a bit of fun mapping the ports. There are apparently two hubs. Once I found the address of one, I was able to turn it on and off with both an iPhone and Lenovo Tab. Both times they stopped charging. The slight complication is the "charging port" which continues to run power even if all ports on both hubs are turned off. I suspect that's the way it's supposed to work, though.

Glad to answer any further questions. I'm doing some interesting things with your software. Thanks!

mvp, 2019-11-12:
Thanks, I will add this device to the list

closed 2019-11-12 by mvp

timocharis, 2019-11-14:
I should have included this information in the first place; sorry. The Dell U2415 has 5 ports, is USB 3.0, and was first issued in late 2014.

Sorry for the complication.

mvp, 2019-11-15:
Thanks! Fixed it in 821cdb8833.