mvp/uhubctl #192: DUB-H7 rev D - powered off, locks up when powering on

tinnocente, 2019-11-12:
Hi all, I've been struggling to get this tool to work reliably for me. My terminal locked up when trying to power a port back on (CTRL+C can't kill it either). Am I doing something wrong? Below were my command attempts:

`root@dev:~# /var/www/usbcontrol/uhubctl
Current status for hub 3-1.2.4, vendor 05e3:0608, 4 ports
   Port 1: 0100 power
   Port 2: 0103 power enable connect [ x ]
   Port 3: 0100 power
   Port 4: 0103 power enable connect [ x ]
Current status for hub 3-1.2, vendor 05e3:0608, 4 ports
   Port 1: 0103 power enable connect [05e0:1200 Symbol Technologies S/N: x Rev x]
   Port 2: 0100 power
   Port 3: 0100 power
   Port 4: 0503 highspeed power enable connect [05e3:0608 USB2.0 Hub]

root@dev:~# /var/www/usbcontrol/uhubctl -a off -p 1 -l 3-1.2

Current status for hub 3-1.2, vendor 05e3:0608, 4 ports
   Port 1: 0103 power enable connect [05e0:1200 Symbol Technologies S/N: x Rev: x]
Sent power off request
New status for hub 3-1.2, vendor 05e3:0608, 4 ports
   Port 1: 0000 off
root@dev:~# /var/www/usbcontrol/uhubctl -a on -p 1 -l 3-1.2
...frozen

root@dev:~# uname -a
Linux dev 3.16.0-4-686-pae #1 SMP Debian 3.16.39-1+deb8u1 (2017-02-22) i686 GNU/Linux

`

mvp, 2019-11-12:
If it locks up, it is most likely Linux kernel issue - you are using quite old version 3.16.
I am 99% sure it is freezing in this call:

https://github.com/mvp/uhubctl/blob/c220668adce2abcb8c9fd7c017ae8b459f213b48/uhubctl.c#L962-L966

Basically, it is freezing when sending USB control message to the hub by calling kernel USB driver.

Can you try connecting this hub to newer Linux box? kernel 4.x or even better 5.x?

mvp, 2019-11-13:
Also, this seems to be almost identical to issue https://github.com/mvp/uhubctl/issues/42, which was reported by you 2 years ago.

mvp, 2019-11-16:
Closing for inactivity. Please reopen if you can reproduce it on newer Linux kernels.

closed 2019-11-16 by mvp