mvp/uhubctl #196: command line ?

plegrand1, 2019-11-24:
Hello
i'm new with uhubctl and i cant make it works.

here is the result of uhubctl : 

```
Current status for hub 4-1 [8087:0024]
  Port 1: 0503 power highspeed enable connect [03f0:af11 HP Photosmart 6520 series TH43T581PM05XP]
  Port 2: 0303 power lowspeed enable connect [413c:2113 Dell KB216 Wired Keyboard]
  Port 3: 0303 power lowspeed enable connect [413c:3010]
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
Current status for hub 1-1 [8087:0024]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
Current status for hub 6 [1d6b:0003 Linux 5.3.0-2-amd64 xhci-hcd xHCI Host Controller 0000:03:00.0, USB 3.00, 2 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 5 [1d6b:0002 Linux 5.3.0-2-amd64 xhci-hcd xHCI Host Controller 0000:03:00.0]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 3 [1d6b:0003 Linux 5.3.0-2-amd64 xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 2 ports]
  Port 1: 0203 power 5gbps U0 enable connect [152d:0561 JMicron ICY BOX 0123456789CF]
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 2 [1d6b:0002 Linux 5.3.0-2-amd64 xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0100 power
  Port 2: 0100 power
```

If i want to power Off the "JMicron ICY BOX", what should be the command line ,
I tryied this : uhubctl -a off -p 1 -l 3, without success.
Does that means that my hub is not compatible ?

Here is the result of lsusb -t -v
```
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
        |__ Port 1: Dev 3, If 2, Class=Mass Storage, Driver=usb-storage, 480M
            ID 03f0:af11 HP, Inc 
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 480M
            ID 03f0:af11 HP, Inc 
        |__ Port 1: Dev 3, If 3, Class=Vendor Specific Class, Driver=, 480M
            ID 03f0:af11 HP, Inc 
        |__ Port 1: Dev 3, If 1, Class=Printer, Driver=usblp, 480M
            ID 03f0:af11 HP, Inc 
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
            ID 413c:2113 Dell Computer Corp. 
        |__ Port 2: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
            ID 413c:2113 Dell Computer Corp. 
        |__ Port 3: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
            ID 413c:3010 Dell Computer Corp. Optical Wheel Mouse
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
        ID 152d:0561 JMicron Technology Corp. / JMicron USA Technology Corp. JMS551 - Sharkoon SATA QuickPort Duo
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        ID 8087:0024 Intel Corp. Integrated Rate Matching Hub

```

Thanks for your help

mvp, 2019-11-24:
Your device is attached to root USB 3 hub. Linux has a bug that root USB 3 hubs do not advertise BOS descriptors, which is necessary for uhubctl to find USB 3/USB 2 hub pairs - without this, these devices look completely independent. Solution might be to turn off power for the same port number on both USB 2 and USB 3 hubs manually, e.g.:

    uhubctl -a off -p 1 -l 3
    uhubctl -a off -p 1 -l 2

That said, it might still not work if root hub does not implement vbus off support. If your device does turn off but quickly comes back, add -r N, where N is number 100 to 1000. Good luck !

mvp, 2019-11-26:
I assume this answered your question. If not, feel free to reopen.

closed 2019-11-26 by mvp