mvp/uhubctl #197: 16 port hubs?

lexmart, 2019-11-30:
Looking for a 16 port hub. Noticed the list only has max 7 for supported hubs. Any recommendations on hubs to try?

mvp, 2019-11-30:
Most hub vendors use standard 4 or 7 port USB chipsets as basic building block. If they need more ports, they usually daisy chain those to add more ports, e.g. 10 port hub is usually 4+4+4 hubs minus 2 ports lost to daisy chaining. Maximum ports I have seen on any hub was 14, but it didn't support per port power switching. Maximum ports for single hub with power switching I have observed is 7, not counting 10 port hubs which use 3 4-port chips.

If you happen to find hub that supports per port power switching, please let us know.

closed 2019-11-30 by mvp

lexmart, 2019-11-30:
I see. Will I have an issue if I daisy chain two 7-port hubs that support per-port power switching?

mvp, 2019-11-30:
Yes, it will work fine. However, avoid changing power state for port used to connect slave hub.

lexmart, 2019-12-04:
You have any experience with Pluggable USB3-HUB10C2 not working?
It's a 10-port hub. If not, I could try ordering one and let you know if it works.

lexmart, 2019-12-04:
https://www.amazon.com/Plugable-10-Port-SuperSpeed-Adapter-Additional/dp/B00JEPIOOQ

mvp, 2019-12-04:
This hub is actually 3 4-port hubs daisy chained internally (2 ports are lost to daisy chaining). It might work - I don't have it, but in general I don't recommend hubs which use such daisy chaining.
For higher port count I recommend using multiple 7 port hubs (with true 7 port chip, not ones with 2*4 port).