mvp/uhubctl #199: Atom Processor Z36xxx/Z37xxx Series USB issue

acoul, 2019-12-10:
Greetings,

I have an ASRock D1800M mboard  that I am trying to power off/on the speakers which are powered through a usb port. there is no driver associated to the speakers, just the power.  here is the output of the uhubctl:

Current status for hub 1-1.2 [05e3:0608 USB2.0 Hub]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0303 power lowspeed enable connect [413c:301a PixArt Dell MS116 USB Optical Mouse]
  Port 4: 0100 power
Current status for hub 1-1 [8087:07e6]
  Port 1: 0303 power lowspeed enable connect [045e:078c LITEON Technology USB Keyboard]
  Port 2: 0503 power highspeed enable connect [05e3:0608 USB2.0 Hub]
  Port 3: 0100 power
  Port 4: 0100 power

issuing `uhubctl -a off -r 100 -l 1-1.2 -p 3` does switch off the mouse. plugging in & out the mouse remains off:

Current status for hub 1-1.2 [05e3:0608 USB2.0 Hub]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0000 off
  Port 4: 0100 power
Current status for hub 1-1 [8087:07e6]
  Port 1: 0303 power lowspeed enable connect [045e:078c LITEON Technology USB Keyboard]
  Port 2: 0507 power highspeed suspend enable connect [05e3:0608 USB2.0 Hub]
  Port 3: 0100 power
  Port 4: 0100 power

with the power off to the mouse usb port, I removed the mouse and plugged the usb speaker jack to that exact same port, and the usb speakers were powered on. `uhubctl` does show that specific port as powered off though.

Finally, I did the same thing, by disabling the whole USB2.0 Hub 
`uhubctl -a off -l 1-1 -p 2`

Current status for hub 1-1 [8087:07e6]
  Port 1: 0303 power lowspeed enable connect [045e:078c LITEON Technology USB Keyboard]
  Port 2: 0000 off
  Port 3: 0100 power
  Port 4: 0100 power

with the exact same results as in my first attempt

mvp, 2019-12-10:
Some hub vendors save few cents and don't implement proper vbus off, and apparently yours is one of those.

Your mouse stays off because data connection is disabled on chip level, but not vbus (5v to USB bus) - this is why speakers are not powered off.

Note that all hubs included in supported list should support vbus off - you can attach any of them to your board and control power on them.

closed 2019-12-10 by mvp

acoul, 2019-12-10:
<b>bummer</b>. and to consider that this bord is supposed to be a low power green eco-friendly board ... thank you for the clarification

acoul, 2019-12-14:
a brief follow-up on this closed topic FWIW

just got an early Christmas present to myself, an NUC6CAYH (J3455) box running void-linux (musl libc). the system feels a bit slower as the core clock is lower, but the 4-cores compared to previous 2, along with the better GPU & the incredible tiny footprint made the day.  it was just a plug & play process simply moving the SSD from the old ASRock D1800M to the new NUC. 

the system USB reports as:
> Celeron N3350/Pentium N4200/Atom E3900 Series USB xHCI

and the `uhubctl` output:
> No compatible smart hubs detected!

Since I was afraid of that outcome, I got myself a usb hub with a switch button. it's difficult to find a usb hub from the `uhubctl` compatibility list on the local current market, especially a USB 3.x. 

So I tried `uhubctl` on it:

> Current status for hub 2-1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power

and then turned-off all the ports <b>(selecting either hub 1-1 or 2-1 would switch both hub ports)</b> `uhubctl -a off  -r 100 -l 2-1 -p all`

> Current status for hub 2-1 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off

unfortunately as in the first post, speakers remained powered on. at least, that usb switch button on the hub, saves the plug/unplug usb speaker process.