mvp/uhubctl #200: Port with Jabra Link 360 audio BT dongle won't power off

bushchannelstepper, 2019-12-14:
Hi,
Thanks for this great utility.
I can switch ports on and off fine with the D-Link DUB-H7 D2 hub using uhubctl.
Except for the Jabra Link 360 I have plugged into the hub.  It accepts the power off command but the port remains alive, unlike anything else I've plugged in.
If I remove the Jabra dongle from the port, that port will switch off ok.  If I plug it into a different port, that port won't switch off with the dongle in, while it would before, or after.
It seems the dongle can override the port/bus/hub when the power off command is sent.
Any thoughts?
Thanks

bushchannelstepper, 2019-12-14:
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
root@walrus:~# uhubctl
Current status for hub 2-1.7.4 [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [1415:0003 SONY SCEH-0036 ALR001LZY3]
  Port 2: 0000 off
  Port 3: 0103 power enable connect [0b0e:a345 Jabra LINK 360 501AA5FF74D5011800]
  Port 4: 0303 power lowspeed enable connect [0463:ffff EATON Ellipse PRO P349J47JG1]
Current status for hub 2-1.7 [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [046d:c52b Logitech USB Receiver]
  Port 2: 0103 power enable connect [0a12:0001]
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 2-1 [8087:0024]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0503 power highspeed enable connect [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 8: 0100 power
Current status for hub 4 [1d6b:0003 Linux 5.3.10-1-pve xhci-hcd xHCI Host Controller 0000:07:00.0, USB 3.00, 2 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 3 [1d6b:0002 Linux 5.3.10-1-pve xhci-hcd xHCI Host Controller 0000:07:00.0]
  Port 1: 0100 power
  Port 2: 0100 power
root@walrus:~# uhubctl -l 2-1.7.4 -p 3 -a off
Current status for hub 2-1.7.4 [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0103 power enable connect [0b0e:a345 Jabra LINK 360 501AA5FF74D5011800]
Sent power off request
New status for hub 2-1.7.4 [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off
root@walrus:~# uhubctl
Current status for hub 2-1.7.4 [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [1415:0003 SONY SCEH-0036 ALR001LZY3]
  Port 2: 0000 off
  Port 3: 0103 power enable connect [0b0e:a345 Jabra LINK 360 501AA5FF74D5011800]
  Port 4: 0303 power lowspeed enable connect [0463:ffff EATON Ellipse PRO P349J47JG1]
Current status for hub 2-1.7 [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [046d:c52b Logitech USB Receiver]
  Port 2: 0103 power enable connect [0a12:0001]
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 2-1 [8087:0024]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0503 power highspeed enable connect [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 8: 0100 power
Current status for hub 4 [1d6b:0003 Linux 5.3.10-1-pve xhci-hcd xHCI Host Controller 0000:07:00.0, USB 3.00, 2 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 3 [1d6b:0002 Linux 5.3.10-1-pve xhci-hcd xHCI Host Controller 0000:07:00.0]
  Port 1: 0100 power
  Port 2: 0100 power
root@walrus:~#

mvp, 2019-12-14:
This sounds very similar to FAQ item https://github.com/mvp/uhubctl/blob/master/README.md#power-comes-back-on-after-few-seconds-on-linux
Can you try using very high value for -r?
Perhaps as high 10000 or 100000, and see if your headset gets turned off for that time?

bushchannelstepper, 2019-12-19:
Thanks for that.  -r 100 indeed shuts it off.  Apologies for missing that in the doco.

closed 2019-12-19 by bushchannelstepper