mvp/uhubctl #201: AmazonBasics HU3641V1 not working on macOS Catalina

glumb, 2019-12-18:
OS: macOs: 10.15.2
libusb: stable 1.0.23

The Hub is connected via a thunderbolt 3 dock.
See command output:
<details><summary> lsusb -v    (click)</summary>

```terminal
2019-12-18 18:13:13.828 system_profiler[24734:663527] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-12-18 18:13:13.829 system_profiler[24734:663527] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-12-18 18:13:13.829 system_profiler[24734:663527] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-12-18 18:13:13.829 system_profiler[24734:663527] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-12-18 18:13:13.832 system_profiler[24734:663527] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-12-18 18:13:13.833 system_profiler[24734:663527] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCITR
      PCI Device ID: 0x15ec 
      PCI Revision ID: 0x0006 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x00 

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCITR
      PCI Device ID: 0x15ec 
      PCI Revision ID: 0x0006 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x01 

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCITR
      PCI Device ID: 0x15f0 
      PCI Revision ID: 0x0006 
      PCI Vendor ID: 0x8086 

        USB3.1 Gen2 Hub:

          Product ID: 0x0625
          Vendor ID: 0x2188  (CalDigit)
          Version: 70.42
          Speed: Up to 10 Gb/s
          Manufacturer: CalDigit, Inc.
          Location ID: 0x02300000 / 1
          Current Available (mA): 900
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            USB3.1 Gen1 Hub:

              Product ID: 0x0620
              Vendor ID: 0x2188  (CalDigit)
              Version: 93.06
              Speed: Up to 5 Gb/s
              Manufacturer: CalDigit, Inc.
              Location ID: 0x02340000 / 4
              Current Available (mA): 900
              Current Required (mA): 0
              Extra Operating Current (mA): 0

                4-Port USB 3.0 Hub:

                  Product ID: 0x0411
                  Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
                  Version: 1.23
                  Speed: Up to 5 Gb/s
                  Manufacturer: Generic
[... 325 lines cut ...]
  Port 1: 0503 power highspeed enable connect []
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect []
  Port 4: 0100 power
Current status for hub 2-4 [2188:0620 CalDigit, Inc. USB3.1 Gen1 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0203 power 5gbps U0 enable connect []
  Port 4: 0203 power 5gbps U0 enable connect []
Current status for hub 2-1 [2188:0610 CalDigit, Inc. USB2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect []
Current status for hub 2-3 [2188:0625 CalDigit, Inc. USB3.1 Gen2 Hub, USB 3.10, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect []
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect []

```
</details>

I don't get correct device locations neither does `lsusb -v` output `Per-port power switching`.
Any help is appreciated 😃

mvp, 2019-12-18:
Which version of libusb you are linking with? You need 1.0.22 or higher.
`brew info libusb` should show it. Also, which version of uhubctl?

Note that `lsusb` is not the same on Mac, it doesn't show output expected, that would be working on Linux.

I find it weird that uhubctl doesn't show device properties in square brackets. Does it work on previous versions of MacOS or on Linux?

mvp, 2019-12-19:
I guess I missed that you are using libusb 1.0.23. I don't have access to Mac with Catalina atm. I should be able to look at it in few days after I return from my trip. Something must be up with either Catalina or libusb 1.0.23...

glumb, 2019-12-22:
I just tried the same hub on another Mac:
OS: macOS Mojave, 10.14.6
ibusb: stable 1.0.23

<details><summary> ubuhctl  (click)</summary>

```terminal
$ uhubctl
Current status for hub 20-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [1e7d:2dcb ROCCAT ROCCAT Kone Pure SEL 000000000000]
Current status for hub 20-5 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
```
</details>

Worked fine, I was able oo switch the ports using `uhubctl -l 20-1 -p 4 -a cycle`.

glumb, 2019-12-22:
Trying another USB C to A adapter actually also worked on Catalina. It looks like the Thunderbolt 3 CallDigit Dock is causing the issues.

glumb, 2019-12-22:
I figured when plugging the Amazon hub in the CallDigit Docks middle USB A port it gets a unique location id and I can finally use uhubctl :)

closed 2019-12-22 by glumb

dsr-seap, 2023-12-12:
> I figured when plugging the Amazon hub in the CallDigit Docks middle USB A port it gets a unique location id and I can finally use uhubctl :)

Thank you, your comment solved a related issue I was having. When using Caldigit TS4 middle USB, I got uhubctl to finally work with TS4 directly. Maybe your hub is not even the TS4 one, but somehow it worked!