mvp/uhubctl #202: uhubclt temporary power off on MacOS

jan-m01, 2019-12-23:
Hello,

thanks for creating this amazing piece of software! I noticed a little issue on MacOS that are only known until now on Linux according to my researches here. Let me explain briefly in which context I'm using uhubctl. 

I'm on MacOS Mojave and I would like to turn off remotely my plugged in external SSDs during a remote desktop session because I'm not able to physically unplug the external SSDs. Im interested in powering them off due to power consumption reasons and to reduce the operation hours of the SSDs. I use the recommended basic USB hub from Amazon with the number HU3641V1.

In the current operation uhubctl is able to shut off the hub ports as expected. But after a few minutes the external SSDs get powered on again. According to my researches here regarding the Linux issue, I tried a variation of the command: uhubctl -a off -r N with different number settings. But unfortunatelly this deactivates the ports only temporary too. 

Is there any fix, tip or solution to this issue? The optimal solution would be for me to use to put my computer to sleep and power off my external SSDs permanently. When I'm accessing it remotely I would like to power the SSDs on again. 

Thanks for your help and merry Christmas

mvp, 2019-12-23:
-r N is only helpful if OS kernel tries to reenable port power back immediately. If it happens few minutes after, it means there is some module/daemon which automatically reenables device much later, supposedly trying to access your SSD again (perhaps it's some SSD maintenance?).

You might be able to fix it by running uhubctl -a off periodically every few minutes.

jan-m01, 2019-12-23:
Thanks for your reply. How do I achieve this with which command? Additionally, when I have to do this my computer cannot go to sleep right?

mvp, 2019-12-23:
Whatever uhubctl command you used to turn port power off, just keep repeating it periodically. I guess you will have to write some script or daemon to do it.

At the end of the day, uhubctl does turn power off, and has no control over what happens few minutes later, if OS decides to turn it back on.

Regarding going to sleep, I'm not sure. I assume when computer is asleep, all processes are frozen. But, there is a possibility that computer going back from sleep will automatically enable all USB ports.

jan-m01, 2019-12-23:
Ok, do you have any idea how I could detect some processes of the OS which try to power on the hubs again?

mvp, 2019-12-23:
I don't know. I have never seen userspace processes to control USB power. If anything, when some userspace process tries to access "missing" device, it typically makes kernel to revive it. Long story short, it could be daunting task to find the culprit.

mvp, 2019-12-31:
Closing for inactivity. Feel free to reopen if you think this is something that can be fixed by uhubctl

closed 2019-12-31 by mvp