mvp/uhubctl #204: Raspberry Pie 4 support?

VSG24, 2019-12-27:
Does this program work under Raspberry Pie 4?

srijansaxena11, 2019-12-28:
+1. Even if it is not completely supported let us know. I currently am using pen drive on one of the USB 3.0 ports. Other 3 ports are available to be used, one of which I have to connect to display/touchscreen which I have to turn off at night by turning off the USB port to which it is connected. Even if both USB 2.0 ports are turned off it is fine.

mvp, 2019-12-29:
Sorry it took so long to respond, I was traveling.
This is ongoing effort tracked in https://github.com/raspberrypi/linux/issues/3079
We still have improper usb descriptors reported by RPi 4, even after using latest beta firmware.
There is a way to hack it in uhubctl, but it is very kludgy - it would still require you to execute uhubctl twice.
Unfortunately, I don't have RPi 4 to provide reliable fix quickly, but this effort will continue.

mvp, 2019-12-31:
@VSG24 , @srijansaxena11 , can you please try compiling from rpi4 branch https://github.com/mvp/uhubctl/tree/rpi4?

You may need need to flash new USB firmware https://www.raspberrypi.org/forums/download/file.php?id=33477&sid=d08e808232df0e2ac10eb1af53932f36
per this article: https://www.raspberrypi.org/forums/viewtopic.php?t=258489

You don't need to use patches mentioned in that article, and should not need to use `-e`.
However, power switching may still require turning off power on all ports, e.g.

To turn off:

    sudo uhubctl -a 0 -l 2

To turn on:

    sudo uhubctl -a 1 -l 2

mvp, 2020-01-03:
Bump: I need someone to try rpi4 fix. I don't have Raspberry Pi 4 (unless you want to send me one 😄 ).
Steps:
1. Flash this USB firmware to your RPi4: https://www.raspberrypi.org/forums/download/file.php?id=33477&sid=d08e808232df0e2ac10eb1af53932f36
2. ```git clone https://github.com/mvp/uhubctl -b rpi4`

mvp, 2020-01-05:
Thanks to @justinasjaronis providing ssh access to RPi4, I think I have a fix https://github.com/mvp/uhubctl/commit/7277ae4ed72bf3b7dfb1de2bf0e1cc3993393748.
Can someone please try building uhubctl from https://github.com/mvp/uhubctl/tree/rpi4, and try it per instructions above? Thanks!

mvp, 2020-01-05:
Fix is merged to master 4aae44ced03980802c5fdadcea16d1bc2f8c0bdb.
It is confirmed working by at least 2 other people.

closed 2020-01-05 by mvp

radokristof, 2020-01-14:
@mvp so it works for all USB ports? The README says that only port 4 will work on an RPi 4.
How can I identify what is the 4th port of the RPi 4?

mvp, 2020-01-14:
Turning off port 4 kills power on all remaining ports.

radokristof, 2020-01-14:
Ok so it doesn't matter which port you use, just need to specify port 4 to cut power?

mvp, 2020-01-14:
Yes. I don't have physical access to rpi4b, you might want to experiment with it yourself :-)

radokristof, 2020-01-14:
Ok, I'll try :) I want to use a project that uses this, so I don't know how it is handled there, but I'll try.