mvp/uhubctl #205: smart hub

marcostrullato, 2020-01-05:
Hi all,

I have a raspi 3b+, and connected to its usb port 2 I have a smart hub (powered). Attached to this hub I have a pendrive I want to turn off.

This is the output of lsusb -t -v

```
:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 13, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 24, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 4: Dev 25, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 3: Dev 8, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
        |__ Port 3: Dev 8, If 1, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 3: Dev 8, If 2, Class=Vendor Specific Class, Driver=usbfs, 480M
        |__ Port 3: Dev 8, If 3, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 3: Dev 8, If 4, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 3: Dev 8, If 5, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 3: Dev 8, If 6, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 4: Dev 9, If 0, Class=Mass Storage, Driver=usb-storage, 480M
```
This is the output of the uhubctl command:
```
sudo uhubctl
Current status for hub 1-1 [0424:9514]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
  Port 2: 0503 power highspeed enable connect [1a40:0101 USB 2.0 Hub]
  Port 3: 0503 power highspeed enable connect [2659:1210 Sundtek MediaTV Pro III (EU) U140603172543]
  Port 4: 0503 power highspeed enable connect [059f:1066 LaCie P'9233 Desktop Drive 0000000025c371df30c3]
  Port 5: 0100 power
```
I have a couple of questions: 
1. is it ok that the hub is not recognized as a new bus?
2. I don't see the pendrive in the output. Is it because my smart hub is not compatible?
3. Which command do you suggest to turn off the port on the smart hub? Shall I use the -e to specify the exact location? If yes, do you have any example? I don't see the option documented...

Thanks

How can I turn off

mvp, 2020-01-05:
1. Hub in itself does not create a bus. Bus is something that all hubs are connected to, possibly in daisy chain.
2. Per FAQ, RPi 3B+ has following USB topology:
> 2 hubs - main hub 1-1, all 4 ports ganged, all controlled by port 2. Second hub 1-1.1 (daisy-chained to main): 3 independently controlled ports, 1 is used for Ethernet+wifi, and ports 2,3 are available with proper per-port power switching. In other words, 2 out of total 4 ports wired outside do support independent power switching on 3B+.

I would expect that your pendrive is connected to hub `1-1.1` (daisy-chained to `1-1`). You should be able to see it in `sudo uhubctl` output. If you don't. we might have an issue to debug.
3. -e is only meaningful for USB3 hubs. Proper command to control should be perhaps
```sudo uhubctl -l 1-1.1 -p 2 -a 0```
where port number 2 is the one that your pendrive is attached to (2 or 3).

marcostrullato, 2020-01-06:
Hi mvp, thanks for following up quickly.

In regards of 2 I read carefully the FAQ and I also went through the closed issues to check if anyone else had the same issue.
Unfortunately the pendrive is not appearing in the outout of sudo uhubctl (I pasted it on my initial message). The same is seen from the lsusb command. Output here and in my initial message.

```
lsusb
Bus 001 Device 009: ID 059f:1066 LaCie, Ltd 
Bus 001 Device 008: ID 2659:1210 Sundtek MediaTV Pro III (EU)
Bus 001 Device 027: ID 0781:5530 SanDisk Corp. Cruzer
Bus 001 Device 026: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 013: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Unfortunately both combinations of the command you suggest do not work.. The output is the same:
```
sudo uhubctl -l 1-1.1 -p 2 -a 0
No compatible smart hubs detected at location 1-1.1!
Run with -h to get usage info
```

If I run "sudo uhubctl -l 1-1 -p 2 -a 0" instead, I get the following output.
```
sudo uhubctl -l 1-1 -p 2 -a 0
Current status for hub 1-1 [0424:9514]
  Port 2: 0503 power highspeed enable connect [1a40:0101 USB 2.0 Hub]
Sent power off request
New status for hub 1-1 [0424:9514]
  Port 2: 0000 off
```
But with this command the port on the raspberry gets turned off.
For a short moment the port get disabled.  I tried also this, forcing the repetitions to 500 with the command "sudo uhubctl -a off -p 2 -r 500". This unfortunately doesn't affect the pendrive which stays active, visible with the lsusb command. 

Reading into the closed issues, this last command might disable the port on the raspberry, however it will then disable also the other three on cascade....and this is something I need to avoid. 
Just to avoid it, I plugged to the raspberry port 2 this additional hub, where I would like to disable that port which.

Shall I open a bug, as the pendrive is not recognized?

Regards

mvp, 2020-01-06:
1. Are you sure you have Raspberry Pi 3B+?:
Raspberry Pi 3 B+ has two internally daisy-chained hubs:
```
$ sudo uhubctl
Current status for hub 1-1.1 [0424:2514, USB 2.00, 3 ports]
  Port 1: 0503 power highspeed enable connect [0424:7800]
  Port 2: 0100 power
  Port 3: 0100 power
Current status for hub 1-1 [0424:2514, USB 2.00, 4 ports]
  Port 1: 0503 power highspeed enable connect [0424:2514, USB 2.00, 3 ports]
  Port 2: 0503 power
  Port 3: 0100 power
  Port 4: 0100 power
```
If you don't see hub `1-1.1`, its either you don't have RPi 3B+, or there is an issue with detecting it.

If you only see hub `1-1`, this is consistent with previous models, e.g. RPi 3B (not plus!).

If your pendrive is connected not directly to RPi, but via additional external hub:
```
Port 2: 0503 power highspeed enable connect [1a40:0101 USB 2.0 Hub]
```
then you can turn off power on it only if that hub `1a40:0101 USB 2.0 Hub` supported per-port power switching. If it does not, it won't work.
You can try disconnecting this hub and plugging pendrive in its place (port 2). In such case, there is high possibility that you would be able to control power on this port.
However, keep in mind that RPi hardware will turn off power for all 4 ports (2 to 5) when you try to control power on port 2, this disconnecting all other devices in your list.

Solution for this would be either get Raspberry Pi 3B+, or getting external hub from supported list.

marcostrullato, 2020-01-07:
I just discovered I have a 3B and not a 3B+ (!) I'm a bit disappointed, I will replace the hardware and try again.

Please close the issue, I will reopen a new one in case.

Regards

mvp, 2020-01-07:
Yeah, RPi 3B+ has best support for power switching across all RPi models (B+,2B,3B,3B+,4B). For 3B+, out of 4 USB ports wired outside, 2 support true per-port power switching. All other models only support turning off all USB ports at once.

closed 2020-01-07 by mvp