mvp/uhubctl #207: working: HAMA - 7 port switching USB 2.0 HUB, self-power

harryzz, 2020-01-14:
from lsusb:
ID 04b4:6560 Cypress Semiconductor Corp. CY7C65640 USB-2.0 "TetraHub"

from uhubctl:
Current status for hub 2-1.4, vendor 04b4:6560, 4 ports
   Port 1: 0103 power enable connect 
   Port 2: 0103 power enable connect 
   Port 3: 0303 lowspeed power enable connect 
   Port 4: 0100 power
Current status for hub 2-1, vendor 04b4:6560, 4 ports
   Port 1: 0100 power
   Port 2: 0103 power enable connect 
   Port 3: 0103 power enable connect 
   Port 4: 0503 highspeed power enable connect [04b4:6560]

regards,
harry

mvp, 2020-01-14:
Please check if VBUS off is supported, follow this FAQ: https://github.com/mvp/uhubctl/blob/master/README.md#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl

We only add devices which properly support VBUS off into supported devices table.

harryzz, 2020-01-14:
Sorry not working in this way.

regards

mvp, 2020-01-14:
No problem, thanks for your report. Please report other hubs if you find anything else compatible and not in the list. Thanks!

closed 2020-01-14 by mvp