mvp/uhubctl #209: future request: command for automatic reconnect device

Corwin616, 2020-01-15:
its possible to find port of connected device by device name or device vid:pid and make "action cycle" for it? including with multiple usb hubs
like this:
```
[root@uhubctl]# ./uhubctl
Current status for hub 1-8 [05e3:0608 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [10c4:ea60 Silicon Labs JXB183 0001]
  Port 2: 0103 power enable connect [0483:5740 STMicroelectronics STM32 Virtual COM Port 56FD66423430]
  Port 3: 0103 power enable connect [1a86:7523 USB2.0-Serial]
  Port 4: 0303 power lowspeed enable connect [0590:0028 OMRON Corporation HHX-CABLE-USB1]
```
 **./uhubctl -a reconnect "0483:5740"**
or
**./uhubctl -a reconnect "Silicon Labs JXB183 0001"**

mvp, 2020-01-16:
You can use simple script like this to get port number of your device:

   PORT=$(uhubctl | grep JXB183 | awk '{print int($2)}'

While you are technically correct that such feature can be supported by uhubctl itself, you can already do it with some minimal wrapper around it.

Also, note that device address (location and port) used by uhubctl are stable - it is not going to change if you turn off/on any devices, as long as you keep your hub chain and end device connected physically to the same ports. This should reduce the need to automatically detect device address - you can put it in your configs without fear that they will randomly change.

Corwin616, 2020-01-16:
thanks for answer! 
its work! :)
but how i can find desired hub?

mabam, 2020-05-21:
This is probably a bit late for you @Corwin616 , but I’ll post this answer anyway for reference.

`uhubctl | awk -F '[: ]' '/USB 2/{u=$5}/JXB183/{print u; print $4}'`
will output hub and port in two lines:
```
1-8
1
```
Or as a variable:
`uhubctlHubPort=$(uhubctl | awk -F '[: ]' '/USB 2/{u=$5}/JXB183/{print u; print $4}')`

To further split hub and port into separate variables, you can use this:
`uhubctlHubPort=$(uhubctl | awk -F '[: ]' '/USB 2/{u=$5}/JXB183/{print u; print $4}'); uhubctlHub=$(printf $uhubctlHubPort); uhubctlPort=$(printf "${uhubctlHubPort##*$'\n'}")`

Corwin616, 2020-06-01:
@mabam thank you!

joeyh, 2020-07-04:
In my experience, the hub's location is not entirely stable. I have a documented case here of a hub moving from 1-1.4 to 4-1.6 and back to 1-1.4 and the only change in the topology of the usb network during that time was plugging in one additional device to the parent hub.

I suspect that power cycling hub could have been what caused the change.

Anyway, even if it were guaranteed to be stable, it would be good for it to be easy to use uhubctl in a way that's robust against random cable rearrangements.

joeyh, 2020-07-04:
Also grepping for the device id doesn't if the port is turned off because uhubctl does not display it then.

Here's something that does, but this feels wrong since uhubctl certianly is looking at the same information.

    uhubctlHub=$(lsusb -tvv | perl -lne 'if (m!/sys/bus/usb/devices/(.*?) !) {$v=$1}; if (/0bc2:3322/) { print $v; last}')

mabam, 2020-07-05:
So with power switched off, while not appearing in uhubctl, the device does appear in lsusb. That is interesting. I would have expected the device to be completely invisible just as if it was unplugged.

Your line (after altering the device id to match one of my devices) produces an empty output here though (running macOS High Sierra).

I’m actually using uhubctl with a launch daemon and sleepwatcher. The launch daemon reads the port and hub at startup and saves it as variables to a temp file. That file is then used for powering off and on the port with sleep and wake. That way I’m fine as long as I only plug my device into a different port while the machine is shut down. But I didn’t encounter my hub moving to a different ID when plugging in additional devices, so for me all works fine.

ganzgustav22, 2020-08-12:
> This is probably a bit late for you @Corwin616 , but I’ll post this answer anyway for reference.
>
> `uhubctl | awk -F '[: ]' '/USB 2/{u=$5}/JXB183/{print u; print $4}'`
> will output hub and port in two [...]

But how does this determine which USB hub/host the device is connected to?

My uhubctl output looks like this:
```
Current status for hub 2 [1d6b:0003 Linux 4.19.118-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 1103 power indicator enable connect [0483:5720 Linux 3.10.14 with dwc2-gadget HID Gadget]
  Port 3: 0303 power lowspeed enable connect [03f0:2b4a Lite-On Technology Corp HP USB Slim Keyboard - Skylab EU]
  Port 4: 0103 power enable connect [058f:511b AllTouch TouchCompose 209.0]
Current status for hub 1 [1d6b:0002 Linux 4.19.118-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]

```

It looks like it's always finding the hub that says "USB 2", not the one that the device is actually connected to. If I change the above line from:
`uhubctl | awk -F '[: ]' '/USB 2/{u=$5}/511b/{print u; print $4}'`

to 
`uhubctl | awk -F '[: ]' '/USB 3/{u=$5}/511b/{print u; print $4}'`

The output suggests that the device 511b (the Alltouch TouchCompose) is connected to the hub with number 2, which is not the case.

mabam, 2020-08-12:
I didn’t think of that. It worked for me as my hardware setting is different.

I’m abroad so I can’t test myself right now, but you might try the following, I hope it works independently from the setting:
`uhubctl | awk -F '[: ]' '/USB [0-9]/{u=$5}/511b/{print u; print $4}'`

mvp, 2020-08-13:
@mabam, in order to switch power off, it is absolutely necessary to turn it off on BOTH USB2 and USB3 hubs (virtual pair).

That said, uhubctl will try to automatically do that for you, whether you specify USB2 or USB3 hub, it should find the other one in pair and do the right thing.
However, sometimes uhubctl can get confused (e.g. check issue #255 when all hubs advertise identical ContainerID), and cannot do that properly.

In such cases, you can use switch `-e` to disable automatic USB2/3 handling, but then you have to identify and turn power off on both USB2 and USB3 hubs separately.

mabam, 2020-08-15:
> I’m abroad so I can’t test myself right now, but you might try the following, I hope it works no matter whether the hub is USB 2 or USB 3:
> `uhubctl | awk -F '[: ]' '/USB [0-9]/{u=$5}/511b/{print u; print $4}'`

I’m back home and have just tested this. It works fine here, though it’s probably “cleaner” to use `uhubctl | awk -F '[: ]' '/USB [23]/{u=$5}/511b/{print u; print $4}'`.

> @mabam, in order to switch power off, it is absolutely necessary to turn it off on BOTH USB2 and USB3 hubs (virtual pair).
>
> That said, uhubctl will try to automatically do that for you, whether you [...]

I read issue #255 back when Roeeklinger60 opened it. I wouldn’t know how to make the awk script work around this bug though.
At least my script should work for those not encountering this bug.

mvp, 2020-12-09:
Implemented in 77aeb7f.

closed 2020-12-09 by mvp

mabam, 2020-12-13:
Just tested it here and it works fine. Good work, thank you!

For my use case, I can’t use the cycle function, though. So I’ll stick to the commands I posted above, implemented in a script in order to save hub and port of my device into a temp file before switching it off. That enables me to switch it on again at a later moment using the info from the temp file.

commit 77aeb7f3 (2020-12-09): Implement searching by device description

Add switch --search (-s) to limit hubs by attached device description.
It is recommended to specify search string that is as unique as possible, e.g. device serial number.
It should work for "off" and "cycle" actions, but not for "on" because device would not be visible on USB.

Closes #209.