mvp/uhubctl #210: Pi3B+ Port 4 supports per port power cycling

virtualhere, 2020-01-20:
Can you update the table so that it shows pi3b+ port 4 supporting the power cycling

mvp, 2020-01-20:
RPi3B+ doesn't have port 4 from USB topology standpoint, and it is incorrect to say that port 4 supports (or doesn't support) power switching. Internal USB topology for all RPi models is already described in [FAQ](https://github.com/mvp/uhubctl/blob/master/README.md#raspberry-pi-turns-power-off-on-all-ports-not-just-the-one-i-specified). In particular, RPi3B+ supports true power switching for 2 ports on hub 1-1.1, and all/nothing power switching for all ports of hub 1-1 via port 2.

closed 2020-01-20 by mvp

mvp, 2020-01-22:
Clarified actual port numbers supported by various RPi models in 232dcc6f6dd2.