mvp/uhubctl #211: uhubctl off doesn't turn off the power to downstream usb device.

qin-zou, 2020-01-21:
I have an usb device that's connected to a hub (USB2517, 0424:2517).
The usb path is 1-1.4 on my linux.
When I issue power off command, ```./uhubctl -a off -p 4 -l 1-1```, nothing happens.
But when I issue power on command, ```./uhubctl -a on -p 4 -l 1-1```, the device will go through a power cycle.
```
Jan 21 19:52:08 user.info kernel: [71611.101665] usb 1-1.4: USB disconnect, device number 20
Jan 21 19:52:08 user.info kernel: [71611.331529] usb 1-1.4: new full-speed USB device number 21 using xhci_hcd
Jan 21 19:52:08 user.info kernel: [71611.338325] usb 1-1.4: new full-speed USB device number 21 using xhci_hcd
```
I would like to have the device stay in the off state, but I can only get it to power cycle.
Any idea what could be the issue?

Here is the information about the usb hub:
```
Bus 001 Device 002: ID 0424:2517
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9
  bDeviceSubClass         0
  bDeviceProtocol         2
  bMaxPacketSize0        64
  idVendor           0x0424
  idProduct          0x2517
  bcdDevice            0.02
  iManufacturer           0
  iProduct                0
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9
      bInterfaceSubClass      0
      bInterfaceProtocol      1
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9
      bInterfaceSubClass      0
      bInterfaceProtocol      2
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             7
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      1 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0103 power enable connect
   Port 3: 0000.0103 power enable connect
   Port 4: 0000.0103 power enable connect
   Port 5: 0000.0103 power enable connect
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            9
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

Thanks.

mvp, 2020-01-21:
Can you try following this FAQ https://github.com/mvp/uhubctl/blob/master/README.md#power-comes-back-on-after-few-seconds-on-linux? Basically, add -r N option

qin-zou, 2020-01-21:
I tried that option already with N=1000, the result is the same.
And my kernel version is 4.9.35.

mvp, 2020-01-21:
Please show output for `sudo uhubctl`

qin-zou, 2020-01-21:
Here is the output:

```
Current status for hub 2 [1d6b:0003 Linux 4.9.35-00005-g10e11352 xhci-hcd xHCI Host Controller 0000:03:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2 [0424:2422]
  Port 1: 0103 power enable connect [10c4:87a0 Silicon Laboratories CP2130 USB-to-SPI Bridge 000EA487]
  Port 2: 0103 power enable connect [10c4:ea90 Silicon Laboratories CP2112 HID USB-to-SMBus Bridge 00AC4D9B]
Current status for hub 1-1 [0424:2517]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [0424:2422]
  Port 3: 0103 power enable connect [10c4:ea90 Silicon Laboratories CP2112 HID USB-to-SMBus Bridge 00AC371D]
  Port 4: 0000 off
  Port 5: 0103 power enable connect [10c4:ea90 Silicon Laboratories CP2112 HID USB-to-SMBus Bridge 00AC2B21]
  Port 6: 0100 power
  Port 7: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.9.35-00005-g10e11352 xhci-hcd xHCI Host Controller 0000:03:00.0]
  Port 1: 0503 power highspeed enable connect [0424:2517]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

This is the lsusb view:
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=hub/7p, 480M
        |__ Port 2: Dev 14, If 0, Class=, Driver=hub/2p, 12M
            |__ Port 1: Dev 17, If 0, Class=, Driver=, 12M
            |__ Port 2: Dev 19, If 0, Class=, Driver=, 12M
        |__ Port 3: Dev 15, If 0, Class=, Driver=, 12M
        |__ Port 4: Dev 21, If 0, Class=, Driver=, 12M
        |__ Port 5: Dev 18, If 0, Class=, Driver=, 12M
```

mvp, 2020-01-21:
Note that your output does not show full details for some hubs, it means you did not run it under `sudo` - it was not able to get all the device properties.
Now, according to your output, hub `1-1` port `4` is turned off.
If your device attached to that port is still powered up, it means this hub does not support VBUS off.

If your device is actually off, but device nodes for it are still present, this is FAQ item https://github.com/mvp/uhubctl#usb-devices-are-not-removed-after-port-power-down-on-linux

qin-zou, 2020-01-22:
I just checked the vbus voltage, the device is actually turned off, but device nodes is still present, just like the FAQ item.
Thanks for the help. Should I go ahead and close this issue?

mvp, 2020-01-22:
Thanks for confirming. I hope this will be fixed in Linux kernel one day...

closed 2020-01-22 by mvp