mvp/uhubctl #212: DLINK DUB-H4/E also works

audryste, 2020-01-30:
DUB-H4/E Works perfectly :)

mvp, 2020-01-30:
Can you please provide more details? The following would be helpful:

* output from uhubctl (can be filtered to this hub)
* full model name as printed on the back
* did you test for VBUS off support? https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl

This is an example of great report https://github.com/mvp/uhubctl/issues/188#issuecomment-553170894

Thanks!

mckenfra, 2020-02-17:
@audryste If you power-off the USB ports (with the power adapter for the hub plugged in), and then unplug the USB lead to the host computer, do the USB ports stay powered off? Or do they power-on again automatically?

mvp, 2020-02-17:
Added in b0d321a.

closed 2020-02-17 by mvp