mvp/uhubctl #213: Power-Off Ubertooth Device on Raspberry 4

csmali, 2020-02-03:
Hi, I was dealing with powering off the ubertooth devices on raspberry 4 and I came across with an issue, here is the problem:

I have an ubertooth on port 1-1.4

[image]

Here is the output of uhubctl command

[image]

I am powering off the device, and red light turns off

[image]

[image]

**Here is the problem**

The part I am not understanding is that : Why 2 yellow led still active although I powered the device off. Also when I plug my mobile phone it keeps charging on that powered off port.

This is the output of lsusb showing that I can per-port switching on 1-1,
[image]

If you help me I would be appreciated . Thanks in advance

mvp, 2020-02-03:
Problem here is that your attempt to turn power off on hub  1-1 doesn't automatically do the same on hub 2 (its dual device). Without this, vbus will not turn off.

Questions:
1. Did you upgrade RPi firmware? vl805 firmware must be 00137ac or later.
2. Did you build uhubctl from master branch? Older binary might not have proper handling for dual devices.
3. Can you please provide container id for RPi 4 USB 2.1 4 port hub? You can find it by looking at `sudo lsusb -v` output, check for container id line.

Workarounds:

It should work if you turn power off/on by 2 commands:

    uhubctl -e -l 1-1 -p 4 -a 0
    uhubctl -e -l 2 -p 4 -a 0

Possible reasons:

  My bet is that container id for dual device is not hardcoded, and might change across models or device batches. I may need to use different hack to make work reliably.

csmali, 2020-02-03:
Here is the vl805 firmware update code I used 

[image]

Output of code update script

[image]

Yes, I used latest master version of uhubctl binary. 

Output of lsusb -v command container ID: 

[image]

I tried what you suggested

[image]

It is  still not working unfortunately

[image]

mvp, 2020-02-03:
Your container id shows correctly.

But your firmware still shows 13701 in your picture. It should be 137ac or more.

If fixing firmware version doesn't help, try turning off all 4 ports. That said, turn off port 4 should kill power to all 4 ports anyway.

csmali, 2020-02-04:
I guess I correctly updated the firmware. But still red led turns off but other 2 yellow led still do not turn off. Since I am not sure, should **vl805** software binary be in a specific folder or something to be able to execute properly?

[image]

mvp, 2020-02-04:
You last picture was showing vl805 firmware as 13701. Does it show newer than that now?

As for using ports separately, it is not possible - it's Raspberry Pi 4 hardware limitation as described in FAQ. Closest you can get to this if you downgrade to Raspberry Pi 3B+ - the only model that supports true per-port power switching on 2 out of 4 ports. However, you lose USB 3 support, and get slower hardware.

Alternatively, you can obtain any external hub from supported list, connect your devices via such hub and control power as you wish.

csmali, 2020-02-04:
Yes I have 137ac now as seen in last picture. I guess I will need external hub. Thanks for your help!

closed 2020-02-04 by csmali