mvp/uhubctl #214: uhubctl port on/off state is out of sync with the kernel hub driver state

samvrlewis, 2020-02-11:
Hi & thanks for the great application!

As far as I can tell, uhubctl bypasses the kernel hub driver when turning off a port by directly sending USB control messages from userspace.

This is fine normally, but I have found that when triggering an overcurrent condition on a hub port, it causes a port (separate to the one with the overcurrent condition) that has been turned off by uhubctl to be turned turned back on. This is seems to be because it is still logically on, as far as the driver is concerned.

The way the kernel deals with overcurrents is shown in the `port_event` function. In this function it calls `hub_power_on`:

https://github.com/torvalds/linux/blob/bb6d3fb354c5ee8d6bde2d576eb7220ea09862b9/drivers/usb/core/hub.c#L5415

In `hub_power_on` each of the ports is powered on, unless the port is logically turned off:

https://github.com/torvalds/linux/blob/bb6d3fb354c5ee8d6bde2d576eb7220ea09862b9/drivers/usb/core/hub.c#L916

The way to logically turn a port off in the driver appears to be the `usb_hub_set_port_power` function:

https://github.com/torvalds/linux/blob/bb6d3fb354c5ee8d6bde2d576eb7220ea09862b9/drivers/usb/core/hub.c#L821

I've traced the driver and turning a port off using something like `echo '0' > /sys/bus/usb/devices/1-1:1.0/1-1-port4/power/pm_qos_no_power_off` _does_ call the `usb_hub_set_port_power` function so that the hub driver is aware the port should be off. Whereas if uhubctl is used to turn off a port, the `usb_hub_set_port_power` function is not called and the hub driver is not aware the port should be turned off. So when an overcurrent event happens, the driver turns ports back on that uhubctl has previously turned off.

I haven't worked through the logic of the driver to know if the hub driver being out of sync with the on/off status of ports might be an issue in ways separate to an over-current condition, but I suppose it could be possible. Especially if the hub driver ever changes.

I don't know how easy this would be to fix, but maybe it would be worth changing uhubctl to use the pm_qos_no_power_off interface? I don't have a deep enough knowledge of how all this works to say if that is feasible, though. In any case I thought it is worth raising this issue and documenting it, in order to save the debugging effort when/if someone else experiences this same issue.

mvp, 2020-02-11:
`uhubctl` does not bypass kernel driver, all the logic is here - it is using standard `libusb_control_transfer`:

https://github.com/mvp/uhubctl/blob/bd92bcef2fb/uhubctl.c#L981-L985

Perhaps kernel USB driver can be improved to intercept such control transfer and call `usb_hub_set_port_power` when needed?
This is mentioned in FAQ https://github.com/mvp/uhubctl#usb-devices-are-not-removed-after-port-power-down-on-linux and discussed at length in linux-usb mailing list https://www.spinics.net/lists/linux-usb/msg157413.html.

Note that this works perfectly in for other operating systems: Darwin, FreeBSD, NetBSD, and never causes this discrepancy.

samvrlewis, 2020-02-11:
Thanks for the response! Sorry, I misunderstood the mechanics of how uhubctl was making the calls to turn off USB ports. That does all make sense, and it is frustrating that linux doesn't currently properly track the state of the ports (and doesn't expose a nice way of shutting off a port from userspace?). Will close this as there doesn't seem to be much that can be done from the uhubctl side.

closed 2020-02-11 by samvrlewis