mvp/uhubctl #215: Definitive Advice for the Raspberry Pi 3B+

lackdaz, 2020-02-12:
Hi! I've been using power switching for the usb ports of the 3B+ for some time but had some time to relook at the latest documentation and found  

> 

Raspberry Pi 3B+
Main hub 1-1, all 4 ports ganged, all controlled by port 2 (turns off secondary hub ports as well):
```
uhubctl -l 1-1 -p 2 -a 0
```
Secondary hub 1-1.1 (daisy-chained to main): 3 independently controlled ports, port 1 is used for Ethernet+wifi, and ports 2,3 support per-port power switching:
```
uhubctl -l 1-1.1 -p 3 -a 0
```
>  

Does this mean that the 3B+ is somehow capable of power switching independently? I vividly remember testing all the 4 ports and I was pretty sure they were ganged together. What are these ports 2, 3 that support per-port power switching?

mvp, 2020-02-12:
Not all ports. But 2 out of 4 yes, only 3B+, not 3B. It's either all 4, or 2 ports on slave hub.

mvp, 2020-02-12:
Note that only 2 ports out of 4 visible on main hub 1-1 are routed outside. 2 more ports on slave hub 1-1.1 are routed outside too.

closed 2020-02-12 by mvp

lackdaz, 2020-02-12:
When you mean the slave hub, do you mean adding a generic usb hub to the pi?

I'm sorry but I'm not super familiar with the port hardware architecture. But are A, B, C, D in the figure below always chained together? 
https://bensherlock.co.uk/wp-content/uploads/2019/04/IMG_20190417_101435-1024x576.jpg

Or is there some way I could switch power on them individually? I have to apologise again for using the issues thread to ask this question. If there is something I could google or read I'd gladly devour this information

mvp, 2020-02-12:
if you run `sudo uhubctl` you should see 2 hubs: 1-1 (main) and 1-1.1 (daisy chained to main). 1-1 has ports 1 to 4. 1 is lost to daisy chaining, 2-3 wired outside, 4 not wired. On slave hub 1-1.1 there are 3 ports, 1 is for networking, 2-3 wired outside, and do support per-port power switching.

lackdaz, 2020-02-12:
hmm, if my understanding is right. This is my `sudo uhubctl` output:

```
Current status for hub 1-1.1 [0424:2514, USB 2.00, 3 ports]
  Port 1: 0503 power highspeed enable connect [0424:7800]
  Port 2: 0103 power enable connect [1d50:6029 marlinfw.org Marlin USB Device 11002007AF3C8CA85C70BD5BF50020C4]
  Port 3: 0100 power
Current status for hub 1-1 [0424:2514, USB 2.00, 4 ports]
  Port 1: 0503 power highspeed enable connect [0424:2514, USB 2.00, 3 ports]
  Port 2: 0503 power highspeed enable connect [1871:0142 AVEO Technology Corp. USB2.0 Camera]
  Port 3: 0100 power
  Port 4: 0100 power
```
I tried to turn off my printer controller board (1-1.1 port 2 of the slave hub? which also corresponds to port A of the diagram I've shown above) by doing `sudo uhubctl -l 1-1.1 -p 2 -a 0 -r 100`. 

This is now my current `sudo uhubctl` output:
```
Current status for hub 1-1.1 [0424:2514, USB 2.00, 3 ports]
  Port 1: 0503 power highspeed enable connect [0424:7800]
  Port 2: 0000 off
  Port 3: 0000 power
Current status for hub 1-1 [0424:2514, USB 2.00, 4 ports]
  Port 1: 0503 power highspeed enable connect [0424:2514, USB 2.00, 3 ports]
  Port 2: 0503 power highspeed enable connect [1871:0142 AVEO Technology Corp. USB2.0 Camera]
  Port 3: 0100 power
  Port 4: 0100 power
```
But my board still remains on. Any idea what is going on?

mvp, 2020-02-12:
Hmm. Documented behavior was reported before by @OutsourcedGuru. Perhaps he was wrong.

I myself don't own RPi 3B+ so didn't really test it. Can you play with this a bit, say turn off both ports 2-3 on slave hub? If that doesn't help, perhaps try port 2 on main hub - that must be a nuclear option, right?

lackdaz, 2020-02-12:
Yeah I've been using the nuclear option for quite some time. Just looked back and saw something shiny. Do you have any recommendations which are the actual pis with individual port switching - in your experience? 

It is *very* convenient to be actually be able to restart my peripherals on Octoprint without having to resort to GPIO.

mvp, 2020-02-12:
Well, before your report, I believed that RPi 3B+ was the best option. Now, I frankly don't understand why Raspberry Pi foundation bothered to introduce this dual hub design over simpler 3B or older single hub layout.

To solve your problem, I recommend getting external hub and connecting printer via this hub. I personally use older D-Link DUB-H7 rev A, works really well, has led light for port on/off state.

lackdaz, 2020-02-13:
Thank you for the recommendations. I'll get one in to try it out!

lackdaz, 2020-05-19:
Hi! I just wanted to check for a definitive way of looking out for a Rev A, Rev D without landing on a B or C.

Do we refer to the H/W version?
[https://imgur.com/UX36SHP](https://imgur.com/UX36SHP)

I'm guessing the B is purple

mvp, 2020-05-19:
Yes, you need to look for H/W ver or rev string. Yours is B1, this one does not work. Look for A or D.