mvp/uhubctl #216: AmazonBasics HU3641V1 - Ports power-up again after shutdown/sleep

mckenfra, 2020-02-14:
Hi,

First of all, your tool is amazing, thank you so much!

I just have one problem, but I'm not sure if there's any way to fix it. It's most likely not an issue with your tool, but I thought I'd ask anyway, as you prob know a lot about these USB hubs after having created this tool!

My problem is that although I can power down the ports on my hub perfectly using your tool, when I power down the host computer, the hub "forgets" the settings. Although the ports remain powered-off initially immediately after host shutdown/sleep, after about 5 seconds, the hub powers up all the ports again. Is there any way of persisting the ports' power-off settings after the host computer is shut down? Or preventing the hub from automatically powering up the ports again when it detects the host computer is switched off?

My setup is as follows:

Hub: AmazonBasics 4-Port hub (HU3641V1) - this is a powered hub, brand new.
Host computer: Mac Mini, macOS High Sierra (hub connected to old USB 2.0 port)
USB peripherals connected to hub: hard disk drives

Ideally, I only want the hub to keep the hard disks running when the Mac Mini is running. I could use your tool to achieve this (with a bit of extra programming to automatically active/deactivate the ports using `uhubctl` during macOS startup/shutdown), but unfortunately the hub seems to do its own thing when the Mac Mini is off.

Do you think it's possible to persist the port-off settings with the AmazonBasics? Or maybe with another hub?

Thanks again!

Francis

mvp, 2020-02-14:
I'm 99.9% sure that this power up is done by MacOS itself as it sends your machine to sleep. I don't know enough about MacOS internals if there is any hook to execute some command at this event. But if it was, solution would be to add uhubctl power off call into this hook/script.

mvp, 2020-02-14:
Another possibility to consider: can you only enable sleep mode, not hibernation or deep sleep? In sleep, it's possible that networking and USB are still continuing to operate (or at least not completely disconnect), and hopefully your USB settings (port state) will "persist".

mckenfra, 2020-02-14:
Thanks for the reply! Yes you are probably right that it's macOS's fault. I've noticed that the seeming delay between putting the Mac to sleep and the hub re-activating itself is actually because the Mac takes a while to fully power-down, after you click "sleep". It seems the instant it fully powers off, the hub comes on again. I'm trying to find a way of programmatically telling macOS to disable the USB port entirely. That should hopefully prevent it sending messages to the hub.

I was just wondering whether the hub itself might have some logic such that when it loses power to the host connection, it automatically resets/restarts itself?

mckenfra, 2020-02-14:
After further testing, I believe this is a feature of the AmazonBasics hub itself. If I use `uhubctl` to turn off all the ports, my hard disks all power down perfectly. But if I then unplug the hub from my Mac Mini, the hard disks attached to the hub all immediately power back up again. I think the AmazonBasics hub must switch to "charging" mode when it detects that the host computer is not providing power. Unfortunately, it appears this "charging" mode overrides any previous port-off settings configured using `uhubctl`.

So I'm going to have to send the hub back. All I want is a powered hub that powers off when the host powers off!

Anyway, this problem has nothing to do with your tool which works really well, thanks a lot. If you or anyone has suggestions about a good hub that persists the power-off setting, let me know! Otherwise, I'll keep searching...

mvp, 2020-02-15:
I have old D-Link DUB-H7 rev A, it doesn't seem to activate its ports unless computer connection is detected. Also, if computer connection is lost, it automatically turns down all ports. However, I don't know how it's is to this old hub these days

closed 2020-02-15 by mvp

arohm-ff, 2020-02-28:
I tested this same behavior on the AmazonBasics HU3641V1 - I believe the behavior you're describing is automatic and dependent on the use of an external power supply with the hub. The external supply is optional for powering the hub, though I don't know if the hub will provide as much power to USB ports without an external supply. In my testing, if the external power supply is not used, the hub will follow the power status of the host.

mckenfra, 2020-02-28:
@arohm-ff Yes, I think you're right, it's most likely an intended feature, because a lot of people want to be able to recharge their gadgets regardless of whether the host computer is on or off.

It just so happens that in my case, this behaviour is exactly the opposite of what I need! When my computer goes to sleep, I want my hard disks to power down, because I hate their constant whirring and humming noise. But I also need the external power supply, because the hard disks don't get enough power from the host computer's USB port. `uhubctl` allowed me to turn off the hard disks, but the problem was that the AmazonBasics hub "forgot" my settings as soon as the host computer disconnected.

Fortunately I took a punt on a super-cheap Belkin hub (F5U404, $10), and it works perfectly - it turns off my hard disks when the host computer goes to sleep. Sadly it's not compatible with `uhubctl`, but you can't have everything...