mvp/uhubctl #218: RPi4 - kernel - internal error

radoslavv, 2020-02-21:
just took latest version of uhubctl and compiled
hub list seems OK:
 
`root@raspberrypi:/home/pi/Packages/uhubctl# ./uhubctl
Current status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect [1058:25e1 Western Digital My Passport 25E1 575851314131383255394355]
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [15a4:9016 Afatech DVB-T 2 010101010600001]
Current status for hub 1 [1d6b:0002 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]`

also POWER OFF command seems to be accepted:
`root@raspberrypi:/home/pi/Packages/uhubctl# ./uhubctl -R -e -l 1-1 -p 4 -a 0
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0503 power highspeed enable connect [15a4:9016 Afatech DVB-T 2 010101010600001]
Sent power off request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off`

but POWER ON causes a kernel error:
`root@raspberrypi:/home/pi/Packages/uhubctl# ./uhubctl -R -e -l 1-1 -p 4 -a 1

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140287] Internal error: Oops: 207 [#1] SMP ARM

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140715] Process kworker/2:1 (pid: 14454, stack limit = 0x0764a7ea)

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140731] Stack: (0xd2145960 to 0xd2146000)

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140751] 5960: bf8fe040 bf8fd02c bf8fd024 bfa2506c 00000000 ff8fd024 00000060 d6050001

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140775] 5980: d2145977 6c691ebf c02b899c efaf12a0 dbc01500 d6052200 d8ac5c00 efaf138c

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140799] 59a0: 00000000 00000001 d21459d4 d21459b8 bfa209ec bf8fc2cc efaf1000 efaf1000

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140822] 59c0: efaf12a0 efaf1004 d2145a3c d21459d8 bfa17090 bfa207e4 bfa2510c c0306c80

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140845] 59e0: 60000013 d6052200 efaf13d4 efaf1000 00000002 c1004d88 bfa168ec bfa16a00

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140868] 5a00: 000d107b 00000000 d2145a3c 6c691ebf c0714914 d6052220 ffffffed d8ac6078

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140891] 5a20: d8ac6000 bfa250b4 d6052200 bfa2317c d2145a74 d2145a40 c07926b4 bfa16aec

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140915] 5a40: c045dde0 c045dce0 d2145a74 c1153ec0 d6052220 00000000 c1153ec4 00000000

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140938] 5a60: bfa250b4 00000093 d2145aa4 d2145a78 c0708554 c07925b0 00000000 d6052220

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140961] 5a80: bfa250b4 d6052220 c0708a38 00000001 c1153e9c 00000000 d2145ae4 d2145aa8

Message from syslogd@raspberrypi at Feb 21 22:52:15 ...
 kernel:[ 8862.140984] 5aa0: c07087c0 c0708324 d2145acc d2145ab8 c0791cac c0791bf4 d6052200 bfa250b4

[... 240 lines cut ...]
[ 8862.142325] [<c0708afc>] (device_initial_probe) from [<c0707558>] (bus_probe_device+0x94/0x9c)
[ 8862.142353] [<c0707558>] (bus_probe_device) from [<c0704fe4>] (device_add+0x468/0x600)
[ 8862.142380] [<c0704fe4>] (device_add) from [<c0790284>] (usb_set_configuration+0x51c/0xa5c)
[ 8862.142408] [<c0790284>] (usb_set_configuration) from [<c079bc98>] (generic_probe+0x3c/0x88)
[ 8862.142434] [<c079bc98>] (generic_probe) from [<c079258c>] (usb_probe_device+0x34/0x4c)
[ 8862.142462] [<c079258c>] (usb_probe_device) from [<c0708554>] (really_probe+0x23c/0x2d4)
[ 8862.142490] [<c0708554>] (really_probe) from [<c07087c0>] (driver_probe_device+0x70/0x1ac)
[ 8862.142518] [<c07087c0>] (driver_probe_device) from [<c0708ad0>] (__device_attach_driver+0x98/0xa8)
[ 8862.142547] [<c0708ad0>] (__device_attach_driver) from [<c0706384>] (bus_for_each_drv+0x88/0xcc)
[ 8862.142574] [<c0706384>] (bus_for_each_drv) from [<c0708294>] (__device_attach+0xe0/0x144)
[ 8862.142602] [<c0708294>] (__device_attach) from [<c0708afc>] (device_initial_probe+0x1c/0x20)
[ 8862.142630] [<c0708afc>] (device_initial_probe) from [<c0707558>] (bus_probe_device+0x94/0x9c)
[ 8862.142658] [<c0707558>] (bus_probe_device) from [<c0704fe4>] (device_add+0x468/0x600)
[ 8862.142686] [<c0704fe4>] (device_add) from [<c0786558>] (usb_new_device+0x27c/0x450)
[ 8862.142715] [<c0786558>] (usb_new_device) from [<c078806c>] (hub_event+0xdc0/0x11d8)
[ 8862.142746] [<c078806c>] (hub_event) from [<c023e028>] (process_one_work+0x170/0x458)
[ 8862.142775] [<c023e028>] (process_one_work) from [<c023e36c>] (worker_thread+0x5c/0x5a4)
[ 8862.142801] [<c023e36c>] (worker_thread) from [<c02446a0>] (kthread+0x138/0x168)
[ 8862.142828] [<c02446a0>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[ 8862.142849] Exception stack(0xd2145fb0 to 0xd2145ff8)
[ 8862.142867] 5fa0:                                     00000000 00000000 00000000 00000000
[ 8862.142891] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 8862.142914] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 8862.142937] Code: e12fff33 e594302c e1a00004 e3a01004 (e5d32010)
[ 8862.143035] ---[ end trace 393a39da52e32e42 ]---`

mvp, 2020-02-21:
If process cannot be killed, it means it did not return from kernel syscall.
I'm 99% sure it is spinning in this `libusb_control_transfer` call:
https://github.com/mvp/uhubctl/blob/b0d321a041a52ad13790c22a91eceec165d9a43b/uhubctl.c#L981-L985
which basically calls libusb, and it in turn calls kernel USB driver to perform control transfer.
If kernel driver is buggy, it may hang there forever and not return from syscall. In that case, you would not be able to any user process that is currently in syscall (e.g. uhubctl).

Which Linux distribution you are running? I would recommend using stable Raspbian.

Can you try using this command instead?
```
uhubctl -l 1-1 -a 0
```
which means turn off for both USB2 and USB3 busses (no `-e`), and all 4 ports. Using only port 4 should turn off all ports anyway, so it should be equivalent. However, turning off port 4 may confuse kernel on ports 1,2,3 because hardware will turn those ports off implicitly.

mvp, 2020-02-26:
This must be Linux kernel bug, or your particular hardware issue.
Please feel free to reopen if you have any new findings and/or details that can help to debug.
So far, no one else has reported any crashes on any Raspberry Pi.

mvp, 2020-02-26:
Another thought - did you upgrade VLI805 firmware to the latest version?
That might be critical to fix your issue.

closed 2020-02-26 by mvp