mvp/uhubctl #220: Incorrect USB2/USB3 device correlation

sre, 2020-02-25:
Hi,

Connecting an AmazonBasics HU3641V1 to a Thinkpad Dock works as expected:

```
earth# uhubctl | grep "Current status"
Current status for hub 2-3.3 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 2-3 [17ef:1010 LENOVO Lenovo ThinkPad Dock, USB 3.00, 4 ports]
Current status for hub 1-3.4 [17ef:100f Lenovo Lenovo ThinkPad Dock Rev1.2]
Current status for hub 1-3.3 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 1-3 [17ef:1010 LENOVO Lenovo ThinkPad Dock, USB 2.10, 4 ports]
earth# uhubctl --vendor 2109 | grep "Current status"
Current status for hub 2-3.3 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 1-3.3 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
earth# uhubctl --vendor 2109:2811 | grep "Current status"
Current status for hub 2-3.3 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 1-3.3 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
```

But if the same hub is connected directly to the laptop, the vendor selection does weird things. It looks like uhubctl assumes The Thinkpad dock to be the USB3 variant of the Amazon hub.

```
earth# uhubctl | grep "Current status"
Current status for hub 2-3 [17ef:1010 LENOVO Lenovo ThinkPad Dock, USB 3.00, 4 ports]
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 1-3.4 [17ef:100f Lenovo Lenovo ThinkPad Dock Rev1.2]
Current status for hub 1-3 [17ef:1010 LENOVO Lenovo ThinkPad Dock, USB 2.10, 4 ports]
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
earth# uhubctl --vendor 2109 | grep "Current status"
Current status for hub 2-3 [17ef:1010 LENOVO Lenovo ThinkPad Dock, USB 3.00, 4 ports]
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 1-3 [17ef:1010 LENOVO Lenovo ThinkPad Dock, USB 2.10, 4 ports]
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
earth# uhubctl --vendor 2109 -e | grep "Current status"
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
```

mvp, 2020-02-25:
Can you please send output of `sudo lsusb -v`, you can filter output only to hubs in question, must importantly I'm looking for ContainerID strings. Thanks!

sre, 2020-02-25:
```
Bus 002 Device 019: ID 17ef:1010 Lenovo Lenovo ThinkPad Dock   
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x17ef Lenovo
  idProduct          0x1010 
  bcdDevice           50.41
  iManufacturer           1 LENOVO                 
  iProduct                2 Lenovo ThinkPad Dock   
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x000d
    Per-port power switching
    Compound device
    Per-port overcurrent protection
  bPwrOn2PwrGood      100 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.4 micro seconds
  wHubDelay          4004 nano seconds
  DeviceRemovable    0x10
 Hub Port Status:
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
   Port 4: 0000.02a0 5Gbps power Rx.Detect
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
[... 558 lines cut ...]
  bDescriptorType      41
  nNbrPorts            11
  wHubCharacteristic 0x000a
    No power switching (usb 1.0)
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x08 0x00
  PortPwrCtrlMask    0xff 0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
   Port 2: 0000.0100 power
   Port 3: 0000.0503 highspeed power enable connect
   Port 4: 0000.0100 power
   Port 5: 0000.0103 power enable connect
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
   Port 8: 0000.0503 highspeed power enable connect
   Port 9: 0000.0100 power
   Port 10: 0000.0100 power
   Port 11: 0000.0100 power
Device Status:     0x0001
  Self Powered
```

mvp, 2020-02-25:
Thank you, it confirmed my suspicions. AmazonBasics and docking station hubs have exactly the same ContainerID, which confuses the hell out of uhubctl.

According to USB 3 standard, each USB 3 device (as well as it's USB 2.1 compatibility partner) are supposed to advertise unique ContainerID, which should be the same for devices which are part of one group. However, on this case, 2 (actually 4 in total) unrelated devices advertise the same ContainerID.

I will think how to remedy this, but it's going to be rather difficult. For time being, workaround is to use -e :-(

mvp, 2020-12-07:
Fixed by 44f963d00f3. Please reopen if still not working.

closed 2020-12-07 by mvp