mvp/uhubctl #223: error: ‘LIBUSB_DT_SUPERSPEED_HUB’ undeclared

tactmaster, 2020-03-13:
Hello , maybe my fault but....
RPi 4
Linux 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
sudo apt-get install libusb-1.0-0-dev
libusb-1.0-0-dev is already the newest version (2:1.0.22-2).

> $ make
> cc -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"v2.0.0\" uhubctl.c -o uhubctl -Wl,-z,relro -lusb-1.0
> uhubctl.c: In function ‘get_hub_info’:
> uhubctl.c:292:45: error: ‘LIBUSB_DT_SUPERSPEED_HUB’ undeclared (first use in this function); did you mean ‘LIBUSB_SPEED_HIGH’?
>      int desc_type = bcd_usb >= USB_SS_BCD ? LIBUSB_DT_SUPERSPEED_HUB
>                                              ^~~~~~~~~~~~~~~~~~~~~~~~
>                                              LIBUSB_SPEED_HIGH
> uhubctl.c:292:45: note: each undeclared identifier is reported only once for each function it appears in
> uhubctl.c:328:26: warning: implicit declaration of function ‘libusb_get_port_path’; did you mean ‘libusb_get_pollfds’? [-Wimplicit-function-declaration]
>              int pcount = libusb_get_port_path(NULL, dev, port_numbers, MAX_HUB_CHAIN);
>                           ^~~~~~~~~~~~~~~~~~~~
>                           libusb_get_pollfds
> uhubctl.c: In function ‘print_port_status’:
> uhubctl.c:488:21: warning: implicit declaration of function ‘libusb_get_parent’; did you mean ‘libusb_get_pollfds’? [-Wimplicit-function-declaration]
>                  if (libusb_get_parent(udev)      == dev &&
>                      ^~~~~~~~~~~~~~~~~
>                      libusb_get_pollfds
> uhubctl.c:488:50: warning: comparison between pointer and integer
>                  if (libusb_get_parent(udev)      == dev &&
>                                                   ^~
> uhubctl.c:489:21: warning: implicit declaration of function ‘libusb_get_port_number’; did you mean ‘libusb_get_bus_number’? [-Wimplicit-function-declaration]
>                      libusb_get_port_number(udev) == port)
>                      ^~~~~~~~~~~~~~~~~~~~~~
>                      libusb_get_bus_number
> make: *** [Makefile:40: uhubctl] Error 1
>

mvp, 2020-03-14:
It doesn't seem like you are property pulling libusb headers. E.g. LIBUSB_DT_SUPERSPEED_HUB is well known since libusb 1.0.16: http://libusb.sourceforge.net/api-1.0/group__libusb__desc.html

It seems that you have some really old headers somewhere

tactmaster, 2020-03-14:
Great, thanks very much. Yeah about application installed /usr/local/include/libusb-1.0/libusb.h 

Fixed now

closed 2020-03-14 by tactmaster