mvp/uhubctl #227: Can't adress port on chained hup

mkoegel, 2020-03-26:
Hi,
I have a webcam on my DELL U2715H monitors USB hub that doesn't work after wake from sleep.
So I wan't to cycle its power with sleepwatcher.
This is on a MacBook Pro 16" with OSX 10.15.3.

The hubs 6 ports seems to be split in two.

```
uhubctl -el 20-4             
Current status for hub 20-4 [0451:8142 A806006164CB, USB 2.10, 2 ports]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect []
Current status for hub 20-4 [0451:8142 A806007164CB, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect []
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect []
```

lsusb shows this for Bus 020:
```
Bus 020 Device 001: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub 
Bus 020 Device 003: ID 0451:8142 Texas Instruments Hub  Serial: A806007164CB
Bus 020 Device 007: ID 0b0e:0410 GN Netcom Jabra SPEAK 410 USB  Serial: 0023781F37D0x010100
Bus 020 Device 004: ID 0451:8142 Texas Instruments Hub  Serial: A806006164CB
Bus 020 Device 009: ID 0c45:6536 Sonix Technology Co., Ltd. USB  Live camera  Serial: SN0001
Bus 020 Device 002: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub 

```

Device 009 is the webcam I want to cycle.

1. Why am I not getting the device names in the []?
2. Is there a way to use the lsusb device # to adress the device? Would make things a lot easier.
3. How can I restart the complete hub if single port is not possible in that case? 

Thanks for your help! Stay safe!

mvp, 2020-03-26:
Which version of uhubctl (try building from master branch)? Which version of libusb (try 1.0.23 - should be the one installed by brew install libusb)?

It's suspicious that even lsusb shows 2 devices with identical serial number A806006164CB - serial numbers are supposed to be unique. That said, I wouldn't fully trust lsusb from brew on max because it simply parses ioreg output.

Can you try using it on Linux and see if it works better?

To answer your questions:
1. Empty info in brackets is not normal - at least vid:pid should be always displayed, and vendor/product/serial of no permission issues.
2. No. uhubctl is using hub location (address) scheme which is stable - it will not change if device is turned off/on or replugged as long as it is connected to the same physical port and hub chain. lsusb address is not stable - it will keep changing on each reconnection, which is very counterproductive when you try to actually use it for power control.
3. You can either reset all hub ports (-p all, or not specify -p at all). Or, you can reset port for parent USB hub, but only if it in itself supports power switching. By resetting parent hub port, it should cause slave hub to fully reset all connected downstream devices.

mkoegel, 2020-03-26:
uhubctl is HEAD-d576db9 updated a few days ago.
libusb is 1.0.23

The only Linux I have available is a Raspberry PI 3+. Does that help?

-p all is not working in my case

```
uhubctl -a cycle -el 20-4 -p all
Error: changing port state for multiple hubs at once is not supported.
Use -l to limit operation to one hub!
```

This is the chain up to the webcam from lsusb -v:
```
USB 3.1 Bus:

      Host Controller Driver: AppleIntelCNLUSBXHCI
      PCI Device ID: 0xa36d 
      PCI Revision ID: 0x0010 
      PCI Vendor ID: 0x8086 

        USB2.0 Hub             :

          Product ID: 0x2817
          Vendor ID: 0x2109  (VIA Labs, Inc.)
          Version: 90.23
          Speed: Up to 480 Mb/s
          Manufacturer: VIA Labs, Inc.         
          Location ID: 0x14100000 / 1
          Current Available (mA): 500
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            Hub:

              Product ID: 0x8142
              Vendor ID: 0x0451  (Texas Instruments)
              Version: 1.00
              Serial Number: A806007164CB
              Speed: Up to 480 Mb/s
              Location ID: 0x14140000 / 11
              Current Available (mA): 500
              Current Required (mA): 0
              Extra Operating Current (mA): 0

                Jabra SPEAK 410 USB:

                  Product ID: 0x0410
                  Vendor ID: 0x0b0e  (GN Netcom)
                  Version: 1.01
                  Serial Number: 0023781F37D0x010100
                  Speed: Up to 12 Mb/s
                  Location ID: 0x14141000 / 13
                  Current Available (mA): 500
                  Current Required (mA): 500
                  Extra Operating Current (mA): 0

                Hub:

                  Product ID: 0x8142
                  Vendor ID: 0x0451  (Texas Instruments)
                  Version: 1.00
                  Serial Number: A806006164CB
                  Speed: Up to 480 Mb/s
                  Location ID: 0x14144000 / 12
                  Current Available (mA): 500
                  Current Required (mA): 0
                  Extra Operating Current (mA): 0

                    USB  Live camera:

                      Product ID: 0x6536
                      Vendor ID: 0x0c45  (Sonix Technology Co., Ltd.)
                      Version: 1.00
                      Serial Number: SN0001
                      Speed: Up to 480 Mb/s
                      Manufacturer: Sonix Technology Co., Ltd.
                      Location ID: 0x14144200 / 14
                      Current Available (mA): 500
                      Current Required (mA): 500
                      Extra Operating Current (mA): 0

```

mvp, 2020-03-26:
Somehow libusb is returning list of USB devices in which 2 which have identical port path 20-4 - that should not be possible. 
This is why you get an error about inability to control multiple devices.
It may manifest a bug in libusb.
Can you please show output of uhubctl without any parameters?

And try attaching your devices (camera or monitor) to RPi3B+ - Raspbian should be ok to test it.

Also, you almost never need to use -e option. If you have to, it typically means a bug somewhere.

mkoegel, 2020-03-26:
I'll need some time to get that Raspberry going again.

lsusb gives me this error 6x as the first output:

`2020-03-26 10:14:05.275 system_profiler[35811:1022916] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be`
Not sure if it's related.

This is the full uhubctl output:

```
Current status for hub 20-4 [0451:8142 A806006164CB, USB 2.10, 2 ports]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect []
Current status for hub 1-4 [0451:8140, USB 3.00, 2 ports]
  Port 1: 06a0 power Rx.Detect
  Port 2: 06a0 power Rx.Detect
Current status for hub 20-4 [0451:8142 A806007164CB, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect []
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect []
Current status for hub 1-4 [0451:8140, USB 3.00, 4 ports]
  Port 1: 06a0 power Rx.Detect
  Port 2: 06a0 power Rx.Detect
  Port 3: 06a0 power Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect []
Current status for hub 20-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Current status for hub 1-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect []
  Port 4: 0503 power highspeed enable connect []
Current status for hub 1-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect []
  Port 3: 0203 power 5gbps U0 enable connect []
  Port 4: 0203 power 5gbps U0 enable connect []

```

mkoegel, 2020-03-26:
Here is what I get on the Raspberry PI.

```
Current status for hub 1-1.3.4 [0451:8142 A806006164CB, USB 2.10, 2 ports]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [0c45:6536 Sonix Technology Co., Ltd. USB  Live camera SN0001]
Current status for hub 1-1.3 [0451:8142 A806007164CB, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [0b0e:0410 Jabra SPEAK 410 USB 0023781F37D0x010100]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0451:8142 A806006164CB, USB 2.10, 2 ports]
Current status for hub 1-1.1 [0424:2514]
  Port 1: 0503 power highspeed enable connect [0424:7800]
  Port 2: 0100 power
  Port 3: 0100 power
Current status for hub 1-1 [0424:2514]
  Port 1: 0503 power highspeed enable connect [0424:2514]
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [0451:8142 A806007164CB, USB 2.10, 4 ports]
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.66-v7+ dwc_otg_hcd DWC OTG Controller 3f980000.usb]
  Port 1: 0503 power highspeed enable connect [0424:2514]

```
BTW the serial numbers of the 2 hubs are different. It's just hard to spot. 
```
A806006164CB
A806007164CB
      ^
```

mvp, 2020-03-26:
Interesting - so it suggests that on Linux it works fine. It suggests that libusb on Mac is broken. Similar happened couple years ago when libusb 1.0.21 was broken on MacOS Yousemite, was fixed in libusb 1.0.22...

mvp, 2020-03-26:
I can't reproduce anything like that on MacOS 10.15.4 (19E266), uhubctl built from master (v2.1.0-31-gd576db98), and libusb 1.0.23 (installed by brew install libusb).
I do not see empty brackets either. Can you try building uhubctl from master, not using brew?

    git clone https://github.com/mvp/uhubctl
    cd uhubctl
    make
    ./uhubctl

mkoegel, 2020-03-26:
The results are identical to the brew version. Just upgraded to MacOS 10.15.4 now, but that didn't change anything.

mvp, 2020-03-26:
I have deleted libusb-1.0.23 (was installed in Oct 2019), reinstalled it again with brew install libusb and built uhubctl against it. Now I can reproduce empty brackets. It seems that something has changed in libusb-1.0.23 in brew, even if version did not. looking at it...

mvp, 2020-03-26:
This appears to be breaking change https://github.com/Homebrew/homebrew-core/commit/02bb855deac07129e3d25c094ec50464d2c64a32
It was working with libusb installed on Sep 19 2019.
Investigating what is the actual problem here.

mvp, 2020-03-27:
I still could not quite figure out what is wrong with libusb-1.0.23 on catalina, but I found workaround.

1. `brew uninstall libusb --ignore-dependencies`
2. Edit file `/usr/local/Homebrew/Library/Taps/homebrew/homebrew-core/Formula/libusb.rb` by commenting out catalina:
```
diff --git a/Formula/libusb.rb b/Formula/libusb.rb
index 9adf5d4f7..f99150868 100644
--- a/Formula/libusb.rb
+++ b/Formula/libusb.rb
@@ -6,7 +6,7 @@ class Libusb < Formula

   bottle do
     cellar :any
-    sha256 "52d3e0d6005ea2fdff5d2c28b972e33f84885f7b1ff9ede3a5603b4f7c332a3e" => :catalina
+    # sha256 "52d3e0d6005ea2fdff5d2c28b972e33f84885f7b1ff9ede3a5603b4f7c332a3e" => :catalina
     sha256 "31b858cce5431f2298524d4e676191477f1e67ee7dd09d02b356cd1219a53c5a" => :mojave
     sha256 "790457dfed646369ee40a2cf1a67f47bc61dbae123b8a4c4937296ae640ffa30" => :high_sierra
     sha256 "600db569dd82dda3e492e53c8023093d003836329b614cea8064ab68d20aca0d" => :sierra
```
Let me know if you can find a way to force installing mojave bottle without patching this formula.
3. `brew install libusb`
4. Build and install `uhubctl`.

Now everything should work.

In meantime, I will keep looking for a real fix.

mvp, 2020-03-27:
Ok, I found actual reason - libusb is broken on MacOS newer than Mojave https://github.com/libusb/libusb/issues/707
Solution for Mac is to use libusb-1.0.22 or apply workaround above.
Hopefully libusb-1.0.24 will provide a fix, which is mentioned in https://github.com/libusb/libusb/issues/707.
Much thanks for reporting this, I did not realize that it was broken all along because I installed libusb before this issue was manifested.

mvp, 2020-03-27:
I have documented this in e3734e0. Please use workaround mentioned in README.md, it should work for you now.

mkoegel, 2020-03-27:
Thanks for the workaround. It works now as expected.

mvp, 2020-03-31:
Closing since workaround is available and is documented. Will need libusb to actually fix it and remove workaround.

closed 2020-03-31 by mvp