mvp/uhubctl #229: another working AmazonBasics 10-port hub

helmutg, 2020-04-04:
The AmazonBasics 10-port hub HUC9002V1ESL practically works.

uhubctl output for Debian package 2.1.0-1

```
Current status for hub 2-4.4.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-4.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
Current status for hub 2-4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
Current status for hub 1-5 [05e3:0608 USB2.0 Hub]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-4.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [redacted]
  Port 4: 0103 power enable connect [redacted]
Current status for hub 1-4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [redacted]
  Port 2: 0503 power highspeed enable connect [redacted]
  Port 3: 0503 power highspeed enable connect [redacted]
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 1-4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
```
Turning ports off seems to work, but I didn't measure power consumption of disabled ports.
I guess you can just add the product identifier next to the very similar HU9002V1SBL.

mvp, 2020-04-04:
Can you check if connected phone stops charging? Thanks

mvp, 2020-04-04:
Googling around, HUC9002V1ESL seems to be simply EU version of HU9002V1SBL - no visual difference between them.

helmutg, 2020-04-04:
Yes, disabling a port makes a phone stop charging. Thank you for the suggestion. Yes, the device looks exactly the same (which was a major reason for trying it).

commit fa524cf7 (2020-04-04): Add AmazonBasics HUC9002V1ESL as supported device

AmazonBasics HUC9002V1ESL seems to be EU version of HU9002V1SBL and works the same.
Closes #229.
closed 2020-04-04 by mvp