mvp/uhubctl #230: List of tested and not working USB hubs

gnoireaux, 2020-04-05:
There is a list of tested and working hubs. I'd like there to be a list of tested and _not_ working hubs.
Because: it might be useful for people not to have to test themselves a hub if it appears on a list of not working hubs.
But: it might become a long list, a maintenance burden…

For exemple
I tested DELL Ultrasharp U2410 24" LCD Monitor
it has 4 USB 2.0 ports.
released in 2009
It advertises to uhubctl the capability to turn off power but doesn't actually do it.

gnoireaux, 2020-04-05:
Thanks for the nice software!

mvp, 2020-04-05:
Maintaining list of all non-working hubs would be quite a long list.
It is very difficult to estimate number of different USB hub models out there, but we can probably do very rough estimation by looking at Linux USB ids database, filtering by hubs:

```
$ curl http://www.linux-usb.org/usb.ids | grep -i hub | wc -l
289
```

Currently supported uhubctl list is 53 (there are some duplicates because different vendors use the same vid:pid).
It means we would have to maintain list that is 5-10 times longer than list of supported hubs. Instead, I prefer to just have list of tested and working hubs.

closed 2020-04-05 by mvp