mvp/uhubctl #231: LG 38WK95C-W works

aluveitie, 2020-04-14:
```
Current status for hub 97-4 [0451:8142 460D007927F2, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect []
```

mvp, 2020-04-15:
Thank you for reporting! However, to be added to compatibility list, I need to ask you to do some extra tests, if you don't mind:

* Can you please check if VBUS off is supported by this device per https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl?
* Your log shows only USB 2.0 connection was used. Can you please check if it works when connected to USB 3.0 upstream port?
* Your log shows that connected device is just empty brackets:`[]`. The only way to get that bad condition is on MacOS if you use `libusb-1.0.23`. Can you please try this fix for MacOS: https://github.com/mvp/uhubctl#compiling
```
brew uninstall --ignore-dependencies libusb
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/5314f1d/Formula/libusb.rb
```
After you reinstall libusb, `uhubctl` should work properly and never show empty brackets.

aluveitie, 2020-04-15:
I've reinstalled libusb

```
Current status for hub 97-4 [0451:8142 460D007927F2, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [043e:9a39 LG Electronics Inc. USB Controls 910NTJJAR444]
```

I did the VBUS check and my phone stopped charging.
It is connected to an USB 3.0 Type C port, but since I'm using that connection for the DisplayPort signal too it's downgraded to USB 2.

mvp, 2020-04-15:
Much thanks for reporting! Added to compatibility list.

commit aa864285 (2020-04-15): Add LG Electronics 38WK95C-W as supported device

Closes #231.
closed 2020-04-15 by mvp