mvp/uhubctl #236: Non-listed hub works with power on/off: Orico USB-3.0

em-winterschon, 2020-04-28:
Here's one hub that was not in the list which has been tested to work correctly. I tested it while running webcam software to see if it would turn off the video stream, which functioned as expected. 

```
Current status for hub 3-11.1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0103 power enable connect [0572:5309 Synaptics USB-A to 3.5mm Jack Audio Adapter 0000000000000000]
  Port 2: 0103 power enable connect [046d:c52f Logitech USB Receiver]
  Port 3: 0507 power highspeed suspend enable connect [046d:0843 Logitech Webcam C930e 82EFFDCE]
  Port 4: 0103 power enable connect [04d9:8008 HOLTEK USB-HID Keyboard AP0000000003]

» s ./uhubctl -a off --level 3-11.1 -p 3
Current status for hub 3-11.1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0503 power highspeed enable connect [046d:0843 Logitech Webcam C930e 82EFFDCE]
Sent power off request
New status for hub 3-11.1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off

» s ./uhubctl -a on --level 3-11.1 -p 3
Current status for hub 3-11.1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off
Sent power on request
New status for hub 3-11.1 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports]
  Port 3: 0101 power connect [046d:0843]
```

The product itself is branded as `Orico USB-3.0`, though I have it connected to a 15ft powered USB 2.0 extension cable so it does not show up as a 3.0 device to the OS. 

* Product: https://www.amazon.com/gp/product/B07QXB4BWS/
* Tested on CentOS 8.1 x86_64
* Compiled after installing `libusbx-devel-1.0.22-1.el8.x86_64.rpm`

mvp, 2020-04-28:
Thanks for your report. Can you please check if this hub supports VBUS off per FAQ?https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl

> Check for VBUS (voltage) off support: plug a phone, USB light or USB fan into USB port of your hub. Try using uhubctl to turn power off on that port, and check that phone stops charging, USB light stops shining or USB fan stops spinning. If VBUS doesn't turn off, your hub manufacturer did not include circuitry to actually cut power off. Such hub would still work to cut off USB data connection, but it cannot turn off power, and we do not consider this supported device.

mvp, 2020-05-04:
Did you have a chance to test VBUS off support for this hub? We do not add devices which appear to work, but don't support VBUS off. Thanks!

mvp, 2020-05-08:
Sorry, closing for inactivity from submitter. Please feel free to reopen if VBUS off is tested to work. Thanks!

closed 2020-05-08 by mvp