mvp/uhubctl #237: TP-Link UH700 (7-port hub) "partially" works

muelthom, 2020-04-28:
I've tested `uhuctl` with the TP-Link model UH700 hub (VID:PID = `0bda:5411`) and switching works to the extent that the **data line is affected but not the power line** as described [here](https://github.com/mvp/uhubctl#usb-30-duality-note) (?).

```
Current status for hub 20-1.3.1 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 20-1.3 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [046d:c52b Logitech USB Receiver]
```

```
$ uhubctl -l 20-1.3 -p 4 -a cycle

Current status for hub 20-1.3 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0103 power enable connect [046d:c52b Logitech USB Receiver]
Sent power off request
New status for hub 20-1.3 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off

Current status for hub 20-1.3 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0000 off
Sent power on request
New status for hub 20-1.3 [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports]
  Port 4: 0101 power connect [046d:c52b]
```

- Product: [TP-Link UH700](https://www.tp-link.com/de/home-networking/computer-accessory/uh700/)
- Tested on macOS Catalina 10.15.4
- uhubctl installed via homebrew

closed 2020-04-28 by muelthom

mvp, 2020-04-29:
Thanks. We had this hub added in #129 and then removed in #133 because it was proven that VBUS off is not working :-(

darksideX1, 2026-10-16:
**Data point: a newer TP-Link UH700 revision (Genesys controllers). VBUS switching works, but per internal hub chip (two ganged groups), not per port.**

The hub in this issue reports Realtek `0bda:5411`; VBUS-off did not work on it. Ours is a different hardware revision:

- label "USB 3.0 7-Port Hub", serial line ending "UN/5.0" (likely hardware version 5.0)
- bought in Europe, October 2026
- uhubctl 2.5.0, Linux 6.8; `-f` is required (without it: "No compatible devices detected")

**How it is built**

Mapped with `lsusb -t`, uhubctl, and an endpoint moved socket by socket. The case has no socket numbers, so they are counted from the right.

```
 front view, sockets numbered from the right

   [7]    [6]    [5]    [4]       [3]     [2]     [1]
  sub p4 sub p3 sub p2 sub p1   main p4 main p3 main p2
                                 ("max 1.5A" sockets)
    |      |      |      |         |       |       |
 +--+------+------+------+--+   +--+-------+-------+-----+
 | SUB hub chip             |   | MAIN hub chip           |
 | 1-2.1 (USB2)             |   | 1-2 (USB2)              |
 | 2-2.1 (USB3)             |   | 2-2 (USB3)              |
 | own ganged power switch  |   | one ganged switch p1-p4 |
 | (not fed by main p1)     |<--+ p1: no socket, carries  |
 +--------------------------+   |     the sub-hub's data  |
                                +-----------+-------------+
                                            |
                                  host xHCI, bus 1 + bus 2

 Both chips: Genesys 05e3:0610 (USB2) / 05e3:0626 (USB3),
 bcdDevice 6.63, wHubCharacteristic 0x00e0 / 0x0000 (ganged)
```

From the right: sockets 1-3 = main p2-p4, sockets 4-7 = sub p1-p4. Main p1 has no socket.

**What goes down**

| You switch off | Power goes down | USB data goes down |
|---|---|---|
| domain main | sockets 1-3 | sockets 1-7 (4-7 keep power) |
| domain sub | sockets 4-7 | sockets 4-7 |
| one main port | nothing | that socket |
| one sub port | nothing | that socket |

The commands (both the USB2 hub and its USB3 twin, per the duality note):

```
# domain main: all ports of the main hub, incl. p1
uhubctl -f -l 1-2 -a off && uhubctl -f -l 2-2 -a off

# domain sub: all ports of the sub hub
uhubctl -f -l 1-2.1 -a off && uhubctl -f -l 2-2.1 -a off

# one main port, e.g. p3 (socket 2): data only
uhubctl -f -l 1-2 -p 3 -a off && uhubctl -f -l 2-2 -p 3 -a off

# one sub port, e.g. p2 (socket 5): data only
uhubctl -f -l 1-2.1 -p 2 -a off && uhubctl -f -l 2-2.1 -p 2 -a off
```

Locations are from our machine (hub on root port 2); `uhubctl -f` lists yours. Back on: the same with `-a on`, USB3 twin first. Whole hub off: both domain lines.

**Gotcha when using uhubctl directly**

- A per-port off "succeeds" but does not cut power. `-p 2 -a off` reports `Port 2: 0000 off` and the device disconnects, yet VBUS stays on because the rest of the gang is on. No error, no warning.
- To really cut power, switch the whole hub (no `-p`) **and** its USB3 twin. Everything on that domain goes down together.
- Without `-f`, uhubctl refuses the hub because it declares ganged power.

**The tests that prove it**

Test endpoint: an ESP32-S3 powered only by USB. Its firmware prints an uptime counter, so a real VBUS cut shows as a reboot (counter restarts) and a data-only drop leaves it counting.

| Switched off (both halves) | Endpoint in | Uptime | Result |
|---|---|---|---|
| sub p2, 10 s | sub | 10040 → 11044 | data only |
| all sub ports, 10 s | sub | 11295 → 250 | **power cut** |
| main p4, 10 s | main | 1756 → 2760 | data only |
| all main ports, 5 s | main | 3011 → 501 | **power cut** |
| all main ports, 10 s | sub | 752 → 1756 | data only, keeps power |
| main p2-p4 (p1 on), 10 s | main | kept counting | data only |

In the "all sub ports" run, the endpoints on main sockets stayed enumerated. 501 is the first reading about 10 s after boot.

Excerpt (one port, both halves):

```
Current status for hub 1-2.1 [05e3:0610 GenesysLogic USB2.1 Hub, ...]
  Port 2: 0103 power enable connect [303a:1001 ...]
New status for hub 1-2.1 [05e3:0610 GenesysLogic USB2.1 Hub, ...]
  Port 2: 0000 off
```

**Summary for this revision**

- Per-port data disable works on all 7 sockets.
- VBUS switches as two independent domains: the 4 left sockets (all sub ports off) and the 3 right sockets (all main ports off, which also takes the left 4 off the bus while they keep power).
- No per-port VBUS. For per-device power cycling, put one device alone in a domain.

Hope this helps others with the same model. @mvp, if useful, feel free to list this revision (Genesys `05e3:0610`/`0626`) with a note like "two ganged groups, 4+3 sockets; `-f` required".