mvp/uhubctl #238: Seagate Backup+ works with uhubctl

ww2, 2020-05-16:
I've just verified that the 3-port USB hub integrated into my Seagate Backup+ Hub external hard drive supports toggling power to all 3 of the ports listed by uhubctl.  Relevant output from uhubctl, with nothing plugged in to either of the two external USB ports: 

> Current status for hub 20-2.4.4.4 [0bc2:ab44 Seagate Backup+ Hub xxxxxxxxxxxx, USB 2.10, 3 ports]
>   Port 1: 0503 power highspeed enable connect [0bc2:ab38 Seagate Backup+ Hub BK NA8TVEKE]
>   Port 2: 0100 power
>   Port 3: 0100 power

Executing `uhubctl -l 20-2.4.4.4 -p 1 -a off` disconnects the drive itself, `-a on` reconnects it.

The 2 external ports are on the front of the device, and when the drive is standing on edge, from top to bottom the ports are numbered 2 and 3, respectively.

With a phone plugged into the bottom external port, `uhubctl -l 20-2.4.4.4 -p 3 -a off` outputs

> Current status for hub 20-2.4.4.4 [0bc2:ab44 Seagate Backup+ Hub xxxxxxxxxxxx, USB 2.10, 3 ports]
>   Port 3: 0100 power
> Sent power off request
> New status for hub 20-2.4.4.4 [0bc2:ab44 Seagate Backup+ Hub xxxxxxxxxxxx, USB 2.10, 3 ports]
>   Port 3: 0000 off

and stops the phone from charging, and `-a on` starts it charging again; likewise for
port 2.

mvp, 2020-05-16:
Thanks for your report, much appreciated!

closed 2020-05-16 by mvp

myudshlihe, 2025-03-23:
No it not cuts the power. I have two of them. It cuts off data and device is disconnected, but powered on.

commit aaf87acc (2020-05-16): Add Seagate Backup Plus Hub STEL8000100 as supported device

Closes #238.