mvp/uhubctl #239: Windows support now that WSL is available?

mrayinteractive, 2020-05-17:
I was going to try this out but figured I would ask to see if anyone has confirmed it *doesn't* work due to the previous libusb issues on Windows....
(https://docs.microsoft.com/en-us/windows/wsl/install-win10)

mvp, 2020-05-17:
Main reason it doesn't work on Windows is because winusb.sys (which libusb is calling is low level layer) doesn't allow to send control messages directly to the hub. I would be very (and pleasantly) surprised if it worked under wsl2. Please check and report back!

mvp, 2020-05-22:
@mrayinteractive, did you have a chance to test it under WSL2?

mvp, 2020-05-26:
Closing for inactivity. Please feel free to reopen if you find it is working under WSL2.

closed 2020-05-26 by mvp

leetfin, 2022-12-17:
@mvp Would it work under WSL2 now with usbipd? https://www.xda-developers.com/wsl-connect-usb-devices-windows-11/

mvp, 2022-12-17:
I highly doubt that this would work. As far as I can tell, `usbipd` is designed to share USB devices, not hubs. For `uhubctl` to work, we need to have low level access to USB hubs - to be able to send USB control messages to turn power off on hub ports.

That said, `uhubctl` is open source - feel free to try. If it does work, report back!
I don't have Windows computers, have not used it in at least 10 years.