mvp/uhubctl #241: Raspberry Pi 4b, same firmware, different hubs

lackdaz, 2020-05-19:
Hi! I have the luxury of having two Raspberry Pi 4bs (personal and "work") and they seem to have different port mappings:

```
pi@octopi4-1:~ $ sudo uhubctl
Current status for hub 2 [1d6b:0003 Linux 5.4.40-v7l+ xhci-hcd xHCI Host Controller 0000:01:  00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 5.4.40-v7l+ xhci-hcd xHCI Host Controller 0000:01:  00.0, USB 2.00, 1 ports]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
```

```
pi@octopi4:~ $ sudo uhubctl
Current status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01: 00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [1d50:6029 marlinfw.org Marlin USB Device 0B00100FAF1E90C55AABBB7CF50020C6]
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01: 00.0]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
```

I ran a `eeprom-update` to the both of them and did a `apt full upgrade` to be sure, and they both have the following firmware version:
```
pi@octopi4-1:~ $ vcgencmd bootloader_version
Apr 16 2020 18:11: 26
version a5e1b95f320810c69441557c5f5f0a7f2460dfb8 (release)
timestamp 1587057086
```

I'm guessing that the Raspberry Pi Foundation decided to switch the USB hubs sometime in the past year?

lackdaz, 2020-05-19:
I've added the `lsusb -t`. The only difference between the two boards is that I have a 3d printer controller board attached to `octopi4`.

```
pi@octopi4-1:~ $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 1: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 480M
```

```
pi@octopi4:~ $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=Communications, Driver=cdc_acm, 12M
        |__ Port 3: Dev 3, If 1, Class=CDC Data, Driver=cdc_acm, 12M
        |__ Port 3: Dev 3, If 2, Class=Mass Storage, Driver=usb-storage, 12M
```

mvp, 2020-05-20:
This can only happen if first RPi4B of yours does not report proper USB descriptors.
Make sure you upgrade vl805 firmware  to version 00137ad https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md
e.g.
```
# rpi-eeprom-update
...
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
```

If both devices show this version for vl805, can you please post output of `sudo lsusb -v` for Pi that is missing this line:
```
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
```

lackdaz, 2020-05-20:
```
pi@octopi4:~ $ sudo rpi-eeprom-update
BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Thu 16 Apr 2020 05:11: 26 PM UTC (1587057086)
 LATEST: Thu 16 Apr 2020 05:11: 26 PM UTC (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
```

```
pi@octopi4-1:~ $ sudo rpi-eeprom-update
BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Thu 16 Apr 2020 05:11:  26 PM UTC (1587057086)
 LATEST: Thu 16 Apr 2020 05:11:  26 PM UTC (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
```
This `sudo lsusb -v` for the pi without 1-1
https://pastebin.com/raw/CNTBPpGV

mvp, 2020-05-20:
Thanks for the logs. Are you sure you built uhubctl from master on both? uhubctl 2.1.0 installed from Raspbian repo via `sudo apt install uhubctl` will have this problem. I should really release master as new version which has required fix.

lackdaz, 2020-05-20:
I installed `sudo apt install uhubctl` individually on both of them. Otherwise they one is actually cloned from the other. They are both running `2.0.0`

```
pi@octopi4-1:~ $ apt policy uhubctl
uhubctl:
  Installed: 2.0.0-5
  Candidate: 2.0.0-5
  Version table:
 *** 2.0.0-5 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
        100 /var/lib/dpkg/status
```

mvp, 2020-05-20:
It means raspberry pi foundation fixed USB descriptors on newer batch. If you build uhubctl from master branch, it will work on both.

mvp, 2020-05-20:
btw, this is a fix that was committed back in January to fix your issue 4aae44ced.

lackdaz, 2020-05-21:
Ok I fixed it. I verify that usb hub switching indeed works as a all on/off on `1-1` or `2` with no individual port control - but that works for me!

closed 2020-05-21 by lackdaz

mvp, 2020-05-22:
I just tagged new release 2.2.0 which will have this fix, hopefully it will make it into Debian/Raspbian/Ubuntu repositories in few weeks/months.