mvp/uhubctl #246: Are Amazon basics HUC9002V1ESL & HU9002V1ESL the same?

Roeeklinger60, 2020-06-18:
Hey,

I just went over the table and you list HUC9002V1ESL as working, going into Amazon.de website I can see 2 versions, one that comes with a USB-B - USB-A cable listed as HU9002V1ESL and another that comes with a USB-B - USB-C cable that is listed as HUC9002V1ESL.

Are they essentially the same but only come with a different cable? Will connecting a USB-B to USB-A cable work with the HUC9002V1ESL?

If nobody knows about the HU9002V1ESL I can make an order and let you know so you can update the table.

mvp, 2020-06-18:
I assume they are the same. But I don't own either one. Please check and report! Thanks

Roeeklinger60, 2020-06-24:
@mvp I ended up getting the HUC9002V1ESL and it is working great as expected. However, toilver opened an issue here and he had the HU9002V1ESL and it worked the same. 99% sure this is the same hub with different USB-B cables. (one comes with USB-A and one comes with USB-C), maybe the second model should be added to the table as well.

mvp, 2020-06-24:
Much thanks for reporting. I will update the table

commit e1718135 (2020-06-25): Update device table for AmazonBasics 10 port hub HU[C]9002V1{SBL,ESL}

Apparently all following hubs: HU9002V1SBL, HU9002V1ESL, HUC9002V1SBL, HUC9002V1EBL
are almost the same, with C models including USB-C cable.
Update device table to reflect this.

Closes #246.
closed 2020-06-25 by mvp