mvp/uhubctl #247: Power off works for some USB devices but not others

Roeeklinger60, 2020-06-19:
I have a BW-H3038A 10-port hub by Pulwtop, which I suspect is almost the same to the Amazon basics hubs since they look almost the same and use the same chip.

I connected this hub to my machine and I connected an iPhone to it (with a USB power detector), I turn off the power using uhubctl and the iPhone stops charging and the USB power detector show 0A and 5V. All as expected.

Then, I leave everything exactly the same, except I plug the iPhone out and plug a Huwaei e3372 4G USB modem instead, as soon as I do that, the amps go straight to 0.10A, which is the default for this device.

I tried turning the port back on and off, rebooting and even changing an OS from Debian to Arch, but I still get the same result every time.

The port light is off, several different devices all get turned off, but as soon as I plug this modem in the amps go up to 0.10 and it receives power (also it's light is flashing so I know it gets power).

There is no way to truly cut power to this device.

Is this a known issue?
Do previously working reported hubs just turn off the amps or they are supposed to turn off the voltage as well?

SlySven, 2020-06-19:
> ...  I turn off the power using uhubctl and the iPhone stops charging and the USB power detector show 0A and 5V.

That sounds suspicious, if power was being removed completely I would expect the voltage (the *pressure* that is exerted on the elections in the conductors to flow) to also be zero as well as the current (how many of them are flowing)!

On the other hand hubs do need to monitor ports to detect when something is plugged in and the mechanism to do that does involve allowing a small amount of current to be drawn (it may be 100mA - or as you say 0.1A) I just don't remember the exact details - if that current IS taken than the hub knows that something is plugged in and then it and the downstream device starts to negotiate how much it can have. {BTW this is why gimmicks such as USB "Fans" or "Lights" are foolish toys - they may well be dumb devices that do not even have any circuitry connected to the pair of data lines and they just suck power off of the power and ground lines - and the hub and upstream do not get any idea of what they are - OTOH that is why they DO make good "Power Switching" testing tools because either they get power and work or they do not and don't: which is what we want when testing whether the hub actually has the circuitry to control the power [which *is* mandatory according to the specification but which is widely ignored and why the list of "working" devices here is such a small percentage of the total hub devices out there]} :roll_eyes:

Roeeklinger60, 2020-06-19:
OK after some testing you seem to be right. This hub has 1 2.4A port and 9 regular ones, when I test on the 2.4A port it now works fine and the USB detector + the modem are completely off and dead, can't even see the measurement as you suggested.

However, the rest of the 9 ports still behave as I described above. USB doc is showing voltage and 0A and when I plug the modem it gets powered on. Seems like just the 2.4A port works, any idea what is going on?

I have the Amazon basics 10-port hub coming in a few days, can anybody confirm all 10 ports work with power switching?

mvp, 2020-06-21:
I think this is due to hardware permitting some trickle current to still go through - this certainly should not happen if port was really turned off.
Closing this issue as it is not really actionable from uhubctl development perspective.
Please open another issue if you find compatible USB hub. Thanks!

closed 2020-06-21 by mvp