mvp/uhubctl #248: Incorrect correlation of USB2/USB3 with Amazon Basics 10 port hub  HU9002V1ESL

toliver, 2020-06-21:
When using the Amazon Basics 10 port hub  HU9002V1ESL, I have observed that the USB2 and USB3 ports are incorrectly associated, which leads to the ports not turning off correctly.

As it can be seen below, when requestin to turn off port 1 in the 4-2.4 hub, it associates it with the USB2 hub 3-2.4.4, instead of the correct one, that would be 3-2.4

```
./uhubctl -l 4-2.4 -a off -p 1
Current status for hub 4-2.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 4-2.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 00a0 off
Current status for hub 3-2.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
Sent power off request
New status for hub 3-2.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
```

toliver, 2020-06-21:
The output of `./uhubctl` is:

```
Current status for hub 4-2.4.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 4-2.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [1058:25a1]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0203 power 5gbps U0 enable connect [1058:10b8]
  Port 4: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
Current status for hub 4-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
Current status for hub 3-2.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 3-2.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 3-2 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
```

toliver, 2020-06-21:
The output of `sudo lsusb -v` is:

```
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x8087 Intel Corp.
  idProduct          0x8000 
  bcdDevice            0.04
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval              12
Hub Descriptor:
  bLength              11
  bDescriptorType      41
  nNbrPorts             8
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        0 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00 0x00
  PortPwrCtrlMask    0xff 0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
   Port 5: 0000.0100 power
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
[... 3032 lines cut ...]
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00 0x02
  PortPwrCtrlMask    0xff 0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0507 highspeed power suspend enable connect
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
   Port 5: 0000.0103 power enable connect
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
   Port 8: 0000.0100 power
   Port 9: 0000.0100 power
   Port 10: 0000.0100 power
   Port 11: 0000.0103 power enable connect
   Port 12: 0000.0503 highspeed power enable connect
   Port 13: 0000.0100 power
   Port 14: 0000.0100 power
   Port 15: 0000.0100 power
Device Status:     0x0001
  Self Powered
```

mvp, 2020-06-22:
Thanks for very detailed report, @toliver. It is essentially a duplicate for issue #220.
Unfortunately, pull request #249 you submitted cannot fully solve this problem (but will break some currently working devices).
Technically, this is due to device manufacturers violating basic USB 2.1/3 promise about ContainerIDs - they are supposed to be unique, but be the same for 2 devices which are correlated.
In this case, we observe that all 4-port basic building blocks for this 10 port hub have exact same ContainerID (failure on manufacturer's part!).
But, that is also true for some other devices like some Thinkpad Docking stations - they have the same ContainerID.

Naive assumption is that all correlated devices would have the same port path (except for bus) does not hold true for all devices - e.g. RPi4B, or Apple HDMI/USB minidock.

I believe correct solution is to implement correlation completely differently that it is done now.
We should have pool of all devices, then run correlation multiple times, on each round giving priority to different clues, e.g. giving the same port part higher priority, but not making it final decision until we exhausted all other possibilities.

I will try to find some time to code this algorithm in next few days, hopefully it won't take long.

For time being, I recommend using `-e` as a workaround.

mvp, 2020-06-22:
BTW, 242 port hub? It must be `lsusb` bug :-)
I'm glad that `uhubctl` detects it properly as 4 port.

```
Bus 004 Device 074: ID 2109:0817 VIA Labs, Inc. 
  bcdUSB               3.10
  bDeviceClass         9 Hub
  nNbrPorts            242
```

toliver, 2020-06-22:
Thanks @mvp. And yes, that looks like an `lsusb` bug.
In fact every time I re-run `lsusb` it reports a different number of ports, always in the 100s or 200s.

I will close the pull request #249

Roeeklinger60, 2020-06-22:
Hey, I have this same exact hub coming in a few days, can you please confirm that all 10 ports work and not just the 3 fast charging ones? Also, do they really cut the power off completely or just lower the power?

toliver, 2020-06-22:
All 10 ports work.
I haven't  checked the voltage when they are off, but the USB powered hard drive I'm testing it with is powering off completely when I turn the port off.

Roeeklinger60, 2020-06-24:
@toliver I have just received my hub today and tested it with a 4G modem (has a light when there is power) + a USB voltage detector. I went through every port 1 by 1 and I can confirm all 10 ports are fully supported and truly cut the power completely.

This is great, a 10 port hub with 3.1 and a 65W power supply that works reliably, I am going to order 15 of those for my project.

However, I hard to turn off each port with 2 commands as I had the same issue you described, I even tried your PR but it just outputted `No compatible smart hubs detected at location 1-2!` even when hub 1-2 did exist.

What parameters exactly do I pass to the `-e` option to make it work with 1 command?

This is my `uhubctl` output if it helps anyone.

```
Current status for hub 20-5.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-2.4.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-2.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
Current status for hub 20-5.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 20-5 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 1-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]

```

mvp, 2020-06-24:
@Roeeklinger60, you need to use 2 commands like this:

      uhubctl -e -l ${usb2} -p ${port} -a ${action}
      uhubctl -e -l ${usb3} -p ${port} -a ${action}

Hardest part is finding correlating usb2/usb3 hubs. I believe for you it would be e.g. 20-5 vs 1-2, 20-5.4 vs 1-2.4 and so on.

mvp, 2020-06-24:
Thinking again about algorithm how to correlate usb2/3 properly, I couldn't come up with any way to correlate hubs if 2 or more hubs like HU9002V1ESL were connected at the same time, short of manually hunting ports down by connecting/disconnecting USB2 and USB3 test devices into the same physical port :-(

Roeeklinger60, 2020-06-25:
@mvp Yes this is exactly the problem I have, to make things more complicated I have the same 20 devices connected to the same 2 hubs and my USB PCIe card mixes the bus numbers each time I reboot. 

There is basically no way for me to figure out which hub is which.

mvp, 2020-06-26:
@Roeeklinger60, can you please share uhubctl output with 2 AmazonBasics 10-port hubs connected? (feel free to hide info about connected devices).
Basically, I want to understand if there is any clue that would allow us to correlate hubs properly in such contrived case.
Perhaps bus numbers would have some pattern?
If we cannot do it manually, I am not sure if any algorithm would work :-(
I assume you are doing it on MacOS?
Thanks!

Roeeklinger60, 2020-06-27:
@mvp Unfourtenly I will not have access to the second hub for about a week and a half. I do however remember the results from the tests I made and I still have access to 5 powered different 10-port USB hubs in my office right now and they all display similar results, I did a lot of testing (reboots with them plugged in, swapped ports, etc).

I am running this on a Linux machine, but also have access to macOS and can do some tests if you want.

This is what I found:
- If I plug a hub in and out physically, I can easily see both the USB 2.0 and USB 3.0 enumerate at the same time in `dmesg`. This of course also happens during startup, but it's a bit harder to tell when there are numerous hubs and also this doesn't solve the issue for `uhubctl` unless we want to constantly monitor `dmesg` from startup and build some kind of table.
- After a reboot, the bus numbers can mix up, so hub 3-1 can become 2-1 or 4-1, so it is not possible to make a table manually once.

However, I did find a way to reliably match USB 2.0 and USB 3.0 hubs, this is true to my system and I don't knot if it will work for everyone.

It seems that the matching USB 3.0 will always have the next bus number **which is an external USB hub**, meaning that if a USB 2.0 hub is located at 3-1, then it's USB 3.0 will be located at 4-1 if there are no built-in USB hubs between them. if there is a built-in USB hub at 4-1, 5-1 will be the correct USB3.0 hub, if there is a second built-in USB hub at 5-1, it will be on 6-1 and so on. This seems to always true, I tested this with 5 different hubs and did 10-15 reboots and checked manually.

So to reliably figure out the correct USB 3.0 we simply need to:
1. List only the external USB hubs.
2. When turning off the power to a certain USB 2.0, simply turn off the power the next USB 3.0 hub directly after it as well.

The only hard part I can think of is figuring out which hub is "built-in" and which one isn't, but I guess we can just match the next identical vendor instead.

I will attach the `uhubctl` with all my hubs connected in 1-2 days when I have access to them, I can make some more tests if you need anything.

mvp, 2020-06-27:
Thanks, @Roeeklinger60 for your thoughts.

Unfortunately, 1) is not going to work. In USB, you cannot tell apart internal huba vs external. I fell into this trap before, it simply doesn't work.

For 2), it sounds interesting. But it assumes that libusb detects and reports hubs in certain order. I don't know if this is something that can be relied upon on all OS. I want to have universal algorithm that would work on any OS, preferably regardless of hub order.

That said, I am all ears for anything that can be used to tell these guys apart and correlate them properly.

Roeeklinger60, 2020-06-27:
@mvp I don't know if this is true for all the OS as well. I have a mac that I can check if that helps.

mvp, 2020-07-02:
@Roeeklinger60 , still waiting for more detailed list for 2 hubs. Note that this issue is duplicate of #220. Let's continue discussion in issue #220.

mvp, 2020-07-02:
Duplicate of #220.

closed 2020-07-02 by mvp

mvp, 2020-12-07:
Fixed by 44f963d00f36 in #288