mvp/uhubctl #250: D-Link DUB-H7

sl8vz, 2020-06-24:
Hi,

Swith: D-Link DUB-H7
HW ver: D2

I'm having issues with this hub. Don't know if you already had similar reported issues or if it's known to work .
Using two different linux machines (both Ubuntu 18/04) doing a power cycle on one of the switch port ends-up in a kernel crash and system lock-up.

Depending on your answer I'll spend more time providing full logs.

Thanks !

mvp, 2020-06-24:
Can I ask you to try it on Mac and perhaps other Linux host, preferably newer Linux kernel or even different Linux distro. Connected USB devices may matter too. My bet that this is Linux kernel bug, not a hardware issue. This is most likely related to this https://github.com/mvp/uhubctl#usb-devices-are-not-removed-after-port-power-down-on-linux and this https://bit.ly/2JzczjZ

sl8vz, 2020-06-25:
Thanks for the information. I will need some time to try that on different hosts/kernels. Unfortunately I will be difficult for me to try on Mac.
Anyway, the usb device not removed might be an issue as when I power cycle the device I have USB serial running on the device so it might be accessing the device while it's actually powered-off. 
I'll give a try to the manual trigger udev to remove the device.

mvp, 2020-06-25:
I don't think this is actionable on uhubctl part, as it is Linux kernel bug. Please reopen if you think otherwise

closed 2020-06-25 by mvp

sl8vz, 2020-07-03:
Hi,
Actually  after reading the kernel list discussion I don't think it's a kernel bug.
The main issue here is that the kernel is not notified that the device is powered-off so there is no reason to prevent any usage from another application.
In my case I always have a serial console opened on the USB device and doing the power-off via uhubctl will necessarily lead to device IOs on a non-existent device.  (And I do't want to close 
I've finally found a way to notify the whole system to not access the device _before_ doing the power-cycles.

The steps would be (usb switch: 1-2.4, port: 4):
```
sudo bash -c "echo 0 > /sys/bus/usb/devices/1-2.4.4/authorized"
uhubctl -a cycle -l 1-2.4 -p 4
```

Note: this is different from the udevadm trigger  in the FAQ

I can submit a PR for the FAQ if you want.

mvp, 2020-07-03:
Please read full thread [here](https://bit.ly/2JzczjZ) - it is definitely a kernel bug. It is kernel USB driver that actually implements power off event, it should be capable to properly propagate this knowledge across the kernel.

That said, I'm glad you found a way to mitigate this. I can add this Linux specific workaround to uhubctl source code. Too bad that this will make uhubctl to always require sudo :-(

sl8vz, 2020-07-03:
You think it's specific to this hub ?

mvp, 2020-07-03:
No, specific to Linux. Remember, uhubctl works on any OS (except for windows, where control messages cannot be sent to hubs).