mvp/uhubctl #252: F5U701-BLK

outis151, 2020-06-29:
Hi, I bought the listed F5U701-BLK and while it is showing up in uhubctl as a device on the onboard usb hub port, it is not showing up as a hub itself.

The hub is on Port 2 (0507) at hub 1-1 :
```
# ./uhubctl
Current status for hub 2-1 [8087:0024]
  Port 1: 0107 power suspend enable connect [04c5:1330]
  Port 2: 0503 power highspeed enable connect [05e3:0723 Generic USB Storage]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1 [8087:0024]
  Port 1: 0503 power highspeed enable connect [2357:010c Realtek 802.11n NIC 00E04C0001]
  Port 2: 0507 power highspeed suspend enable connect [1a40:0101 USB 2.0 Hub]
  Port 3: 0100 power
  Port 4: 0100 power
```

dmesg output (As you can see it is registering as two 4 port USB hubs):
```
[46992.989303] usb 1-1.2: new high-speed USB device number 6 using ehci-pci
[46993.098333] usb 1-1.2: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[46993.098336] usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[46993.098338] usb 1-1.2: Product: USB 2.0 Hub
[46993.098764] hub 1-1.2:1.0: USB hub found
[46993.098938] hub 1-1.2:1.0: 4 ports detected
[46993.485314] usb 1-1.2.1: new high-speed USB device number 7 using ehci-pci
[46993.693961] usb 1-1.2.1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[46993.693964] usb 1-1.2.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[46993.693966] usb 1-1.2.1: Product: USB 2.0 Hub
[46993.694347] hub 1-1.2.1:1.0: USB hub found
[46993.694532] hub 1-1.2.1:1.0: 4 ports detected
```

lsusb on the device:

```
# lsusb -s 001:006 -v | grep "power"
can't get debug descriptor: Resource temporarily unavailable
    Ganged power switching
   Port 1: 0000.0507 highspeed power suspend enable connect
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power

# lsusb -s 001:007 -v | grep "power"
can't get debug descriptor: Resource temporarily unavailable
    Ganged power switching
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
```

Am I doing something wrong? The HUB seems to support some kind of power switching ("Ganged power switching") and is currently also listed. Thanks

mvp, 2020-06-29:
Currently, uhubctl does not support ganged power switching hubs.

Sorry that information in supported device table mislead you that hub was supported. Historically, we collected information about supported hubs from user reports, but they are not always very reliable. I will remove this hub from supported list :-(

closed 2020-06-29 by mvp

outis151, 2020-06-29:
Ok no problem, thanks for the quick response