mvp/uhubctl #253: StarTech ST4200USBM no VBUS

Shed1990, 2020-07-04:
Hey just got one of these hubs for an RPi project - uhub-ctl stops data on a per-port basis and switches off the LED next to the port, but voltage/power still flows to device.  Any advice would be appreciated if it might work, otherwise you might want to note it as not having VBUS support.  All the best

mvp, 2020-07-05:
Are you 100% sure it doesn't turn power off? @joeyh reported it working properly in #112 a while ago

Shed1990, 2020-07-05:
I'm as sure as I can be - USB LED light stays lit when port is 'off'.  Tried all 4 ports, all the same.  Data stops and indicator light turns off, but there is still power to the device.

mvp, 2020-07-05:
Thanks for reporting, I will remove it from the list :-(

mvp, 2020-07-06:
Removed from table in b8a1b808e7aac693004a.

closed 2020-07-06 by mvp