mvp/uhubctl #254: Is is possible to keep the port alive but not send the power to the port?

pjones112000, 2020-07-06:
I want to essentially switch a port to only data, no power output.  I could get a data only cable or tape over the power pin; however, a more stable and reliable means would be through software.  This utility gets me 90% of the way there; however, (at least with Raspbian and Raspberry Pi) it will completely disable the port rather than just disable power output.  Is this even possible?

mvp, 2020-07-06:
No, it is not possible. USB must have at least some power in order to have data connection. Absolute minimum power is 100mA for USB negotiation phase, and standard requires 500mA for typical USB connection.

closed 2020-07-06 by mvp

pjones112000, 2020-07-06:
Thanks.  The issue I have is that I have a couple of devices that get just enough power from the USB to stay partially active when they are powered off.  I was hoping to be able to be able through software to disable/enable the ports rather than physically disconnecting, powering off my Raspberry Pi or taping over the power pin.  Thanks.

commit b8a1b808 (2020-07-06): Remove StarTech ST4200USBM from supported table

There are multiple reports that VBUS off is not supported by ST4200USBM.
Closes #254.