mvp/uhubctl #255: Multiple HUC9002V1SBL Amazonbasics USB hubs

Roeeklinger60, 2020-07-08:
Finally, I had the chance to plug all 4 of my HUC9002V1SBL to my Linux, I also have a macOS device that I can connect them to if needed.

As requested, here is the output of `uhubctl` and `lsusb -v`, hopefully, this can give some insight for correctly correlating the USB2.0 and USB3.0 hubs.

Currently, as a workaround, I use a wrapper script that lists them by order and assumes the USB3 loads right after the USB2, this works every time and after reboots but I have not tested it on any other machine than my own.

Also worth mentioning - `uhubctl` worked very slow with all 4 hubs connected and would sometimes hang. I disabled usbcore with `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.autosuspend=-1"` and as soon as I did everything started to work smoothly and with no delay. I have no idea why.

`uhubctl:`
```
Current status for hub 8-1 [8087:8000]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
Current status for hub 5-1 [8087:8008]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [24ae:1813 RAPOO Rapoo 2.4G Wireless Device]
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0bda:0129 Generic USB2.0-CRW 20100201396000000]
Current status for hub 12-1.4.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 12-1.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
Current status for hub 12-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
Current status for hub 12 [1d6b:0003 Linux 4.19.0-9-amd64 xhci-hcd xHCI Host Controller 0000:06:00.0, USB 3.00, 2 ports]
  Port 1: 0203 power 5gbps U0 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 11-1.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 11-1.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 11-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 11 [1d6b:0002 Linux 4.19.0-9-amd64 xhci-hcd xHCI Host Controller 0000:06:00.0]
  Port 1: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 2: 0100 power
Current status for hub 9-1.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 9-1.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
Current status for hub 9-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
[... 4177 lines cut ...]
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             2
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
   Port 2: 0000.0100 power
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

mvp, 2020-07-08:
Thanks! Unfortunately, `lsusb -v` doesn't provide full information. Can you please get output of `sudo lsusb -v`?

mvp, 2020-07-08:
And yes, getting uhubctl output with the same hubs on Macos is much appreciated. Note that just by looking at these logs, it's impossible to tell correlation. Can you also manually mark correlating hub numbers (if you found it manually?)

Roeeklinger60, 2020-07-10:
I will attach the macOS output in a few days when I have access to it.

As I said in my first post, I have a method to correlate the USB hubs which works every time, but I am not sure if it only on my Debian machine. It also only works if I can tell which hub is external and which is internal.

1. List the external USB hubs.
2. Sort them by smallest to largest hub number.
3. The USB3.0 will always be on the following line from the USB2.0 hub.

So for the above `uhubctl` output:
hub1 usb2 - 3-1
hub1 usb3 - 4-1
hub2 usb2 - 6-1
hub2 usb3 - 7-1
hub3 usb2 - 9-1
hub3 usb3 - 10-1
hub4 usb2 - 11-1
hub4 usb3 - 12-1

5-1 and 8-1 are internal hubs so they are skipped.

Hopefully this helps.

Here is `sudo lsusb -v`:
```
root@raspberry:/home/pi# sudo lsusb -v

Bus 004 Device 002: ID 8087:8000 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x8087 Intel Corp.
  idProduct          0x8000 
  bcdDevice            0.05
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
[... 5201 lines cut ...]
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             2
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
   Port 2: 0000.0100 power
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

mvp, 2020-07-10:
Thanks a lot, @Roeeklinger60. That output is impossibly long - perhaps next time link to pastebin.com :-)
Its weird that BOS ContainerID still failed to work even with `sudo lsusb -v`.

I have just one StarTech ST4300USB3 4-port USB3 hub (which does not support vbus off), and while your theory seems to be true on Linux, it breaks down on Mac.
I observe this on Macbook Pro, connecting ST4300USB3 to different USB-C ports:

| MBP USB-C port | usb2 | usb3 |
| --- | --- | --- |
| Left port 1 | 20-4 | 0-2 |
| Left port 2 | 20-6 | 0-1 |
| Right port 1 | 20-1 | 1-1 |
| Right port 2 | 20-5 | 1-2 |

Bus and port numbering does not seem to follow any pattern 👎 

When I connect this hub to Linux box, I observe nice predictable numbering, when USB2 and USB3 device busses always differ by 1, e.g.:

| usb2 | usb3 |
| --- | --- |
| 3-3.1 | 4-3.1 |

Looking forward for your output from Mac. I wish I had 2 10-port hubs to play with...

c-MM, 2020-08-12:
At least in my case #261 it would solve the problem if the level would have to be equal for the pair.
Adding the level to the command line doesn't help:
```
root@z2srv:~# ~mm/uhubctl/uhubctl -L 2 -l 2-1 -p 1 -a on
Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
Sent power on request
New status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 1-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
Sent power on request
New status for hub 1-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
```
It still uses the wrong pair.
Another option which would make things more convenient on those broken hubs would be to allow specification of the pair like:
```
uhubctl -e -l 2-1 -l 1-3 -p 1 -a off
````

c-MM, 2020-08-12:
> Looking forward for your output from Mac. I wish I had 2 10-port hubs to play with...

I offer you one, let me know how to get it it you.

mvp, 2020-11-26:
This should be solved by #288, please try it, @c-MM, @Roeeklinger60.
I am closing this as duplicate of #220.

closed 2020-11-26 by mvp